mutability/dump1090 #7: Add a gain-scanning mode

mutability, 2015-02-03:
Count messages for short periods with different gain settings and accumulate the results, display the results in a table, suggest the best gain value to set.

keptenkurk, 2015-02-03:
Would be cool feature. Maybe record furthest seen as well in that table? 
Might be feature for web page / json export anyway: furthest seen last 10 minutes?

mutability, 2015-02-03:
I'm thinking of a scanning hop interval of around 1 second so I don't think position data is going to be reliable (you need 2 messages to get a global position, and they're sent at most 2/second)

The webmap does the distance calculation itself anyway already

keptenkurk, 2015-02-03:
Agree on that. Looking forward to this scanning feature!

Romeo-Golf, 2017-01-09:
This would be great all the folks with tuners that perform better with a gain value other than maximum, has there been any further interest in this functionality?

closed 2019-09-14 by mutability