mutability/dump1090 #24: error in post-installation script during upgrade to 1.14

jburgess777, 2015-02-19:
Upgrading from 1.13 package to to 1.14:
```
pi@raspberrypi ~ $ sudo apt-get install dump1090-mutability
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be upgraded:
  dump1090-mutability
1 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
Need to get 213 kB of archives.
After this operation, 10.2 kB of additional disk space will be used.
Get:1 http://repo.mutability.co.uk/raspbian/ wheezy/rpi dump1090-mutability armhf 1.14 [213 kB]
Fetched 213 kB in 0s (408 kB/s)         
Preconfiguring packages ...
(Reading database ... 74717 files and directories currently installed.)
Preparing to replace dump1090-mutability 1.13 (using .../dump1090-mutability_1.14_armhf.deb) ...
Unpacking replacement dump1090-mutability ...
Setting up dump1090-mutability (1.14) ...

Configuration file `/usr/share/dump1090-mutability/html/script.js'
 ==> File on system created by you or by a script.
 ==> File also in package provided by package maintainer.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** script.js (Y/I/N/O/D/Z) [default=N] ? y
Installing new version of config file /usr/share/dump1090-mutability/html/script.js ...

Configuration file `/usr/share/dump1090-mutability/html/planeObject.js'
 ==> File on system created by you or by a script.
 ==> File also in package provided by package maintainer.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** planeObject.js (Y/I/N/O/D/Z) [default=N] ? y
Installing new version of config file /usr/share/dump1090-mutability/html/planeObject.js ...
Installing new version of config file /etc/init.d/dump1090-mutability ...
adduser: The user `root' already exists. Exiting.
dpkg: error processing dump1090-mutability (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 dump1090-mutability
E: Sub-process /usr/bin/dpkg returned an error code (1)
pi@raspberrypi ~ $ 
```

jburgess777, 2015-02-19:
It looks this is caused because I set DUMP1090_USER="root". The adduser script generally doesn't complain about system users already existing but it insists they have UIDs in a specific range, which doesn't include root:
```
# a user with this name already exists; it's a problem when it's not a system user
my $tmp_u = getpwnam($new_name);
if (($tmp_u >= $config{"first_system_uid"}) and ($tmp_u <= $config{"last_system_uid"})) {
    printf (gtx("The system user `%s' already exists. Exiting.\n"), $new_name) if $verbose;
    exit RET_OK
}
warnf (gtx("The user `%s' already exists. Exiting.\n"), $new_name);
```
some other scripts seem to use getent to check for existence instead
```
if ! getent passwd sshd >/dev/null; then
        adduser --quiet --system --no-create-home --home /var/run/sshd --shell /usr/sbin/nologin sshd
fi
```

mutability, 2015-02-19:
Yeah, all sorts of things are likely to break if you say to use root. Don't use root. No, really, don't use root. It's a terrible idea :)

I'll see what I can sort out in terms of handling existing users better though, thanks for the report.

jburgess777, 2015-02-19:
I knew it was a bad idea but dump1090 had previously been running as root before moving to the mutability package and I wasn't sure that a non-root user would have access to the USB device. 

I have set the user back to dump1090, reinstalled the package and all is OK now.

closed 2015-02-19 by mutability

mutability, 2015-02-19:
Should be sorted for the future - it'll reject root now, and it uses getent as you suggested to avoid trying to create the user if it's already there.

commit ab8c4db8 (2015-02-19): Fix edge cases when specifying the user to run as.

 * check the username fits the format expected by adduser;
 * don't allow root;
 * if the user exists, don't try to create it; usually, this is
   harmless, but it fails if the user exists with a UID outside the
   normal range for system users.

Fixes #24.