mutability/dump1090 #37: decode mode A/C in --oversample mode

hhm0, 2015-02-26:
It would be wonderful to support this! :-)

mutability, 2015-02-26:
No fundamental reason this can't be done, it's just not high on my list at the moment.

mutability, 2016-03-20:
I got as far as deciding that tracking phase in terms of a 60MHz clock seems to be the way to go, but no further at the moment.

mutability, 2016-03-20:
I put together something quick here: https://github.com/mutability/dump1090/tree/modeac
It seems to decode some stuff at least..

Be aware that you will need quite a lot more CPU to process Mode A/C (partly to run the demodulator, partly because there are some scaling problems in the original mode a/c tracking code that I have not looked at yet - the cost goes up with the square of the number of tracked aircraft)

jester-xbmc, 2016-03-20:
Thanks for the work, i'll have a play, more CPU meaning a pi3 is out ?

mutability, 2016-03-20:
It's fine on a Pi 3 (~25% CPU), probably a Pi 2 is OK too (haven't tried), but my Pi 1B+ was running out of CPU.

mutability, 2016-03-20:
Also the demodulator is currently tuned to be quite aggressive about rejecting noise since there's no real way to validate a mode a/c message (no error detection/correction at all). With my test site here it's only decoding 15-20% as many messages as the original 2MHz decoder, but the original decoder has a real problem with junk messages here (half the interactive display is junk with A/C on)

jester-xbmc, 2016-03-20:
Thanks, Understood, I have a spare pi3 on the way so I can do some testing.

jthawkes, 2016-03-20:
How do I compile and install it? I can't seem to git clone it.

mutability, 2016-03-20:
It's a branch:

    $ git clone https://github.com/mutability/dump1090.git -b modeac

jthawkes, 2016-03-20:
That worked.
Thanks. Not too familiar with github, other than having to block it at work for several months.

mutability, 2016-03-21:
This seems stable enough that I think I'll go ahead and apply it to master.

jester-xbmc, 2016-03-21:
look good here, i'd say bump to master, thanks for the effort !

jthawkes, 2016-03-21:
My "Tracks with Single messages" graph increased significantly.
It increased about 6 fold. The same for unique tracks.

mutability, 2016-03-21:
Yes. Mode A/C can't really be error-checked so you will be getting all sorts of noise.

mutability, 2016-03-21:
What I would be interested in is someone to actually look at whether the A/C data is _useful_
(it's not very interesting here because almost everything is Mode S too)

jester-xbmc, 2016-03-21:
Its useful tracking mil aircraft, think B-52

mutability, 2016-03-21:
Sure. How well do the changes work for that?

jester-xbmc, 2016-03-21:
well previous mode a/c was ok, but I havent used it as I run oversampled as default, so bit tricky to compare, always i'm waiting for a "juicy target" ;)

jthawkes, 2016-03-21:
Mode C will still be used in the US and Aus for a while. Not so much for Commercial traffic but definately for GA traffic. The US mandate doesn't come into effect until 2020. Even then, only where Mode C transponders are currently required.

mutability, 2016-03-21:
Yes, I understand all this. What I am asking for is for people in areas where Mode A/C is actually interesting (that's you, guys!) to give me some feedback on whether the 2.4MHz demodulator is working well or not. All I can test here is that I get a bunch of A/C returns that match up with the Mode S returns and a low rate of garbage messages. There are few A/C-only aircraft I can test against here.

jester-xbmc, 2016-03-21:
i'm in the same boat, although I'm trying to use this for mode a/c only aircraft, they are much more common in the US, however lots of US mil aircraft movements here lately so I hope to be able to report something useful soon for you.

jthawkes, 2016-03-21:
Hmm, I thought mode Es were suppose to broadcast mode C for quite a while. I think it is needed for tcas. 
I think it is hard to see. Need others to be sending the info to mlat so we can get a position fix.

mighty3507, 2016-03-22:
Running on sites 6173 and 6232 located in Southern California. CPU up about 5%.
Like jthawkes, Unique and Single tracks increased by 6 to 10 times.
This is a portion of my listing using view1090-mutability --modeac:
~005673 A     5673    
~001145 A     1145    
~006315 A     6315    
 A8E1F6 S ac  3346    
 A4DD02 S  c               
 A84CE2 S  c  3557

Are the A and C only aircraft supposed to show list of tracked aircraft?
--Steve

mutability, 2016-03-22:
This is on master now.

closed 2016-03-22 by mutability

hhm0, 2016-03-22:
Thank you so much! Working well here so far, without any noticeable
error returns.