mutability/dump1090 #49: include rtl_test with your librtl deployment

jester-xbmc, 2015-06-11:
Could you include rtl_test in your librtl deployment ? its mentioned in your install documentation.

If you do a clean install it would help to have this available so you can do ppm offset calculations without having to install and compile it yourself.

mutability, 2015-06-11:
It's already packaged as rtl-sdr?

jester-xbmc, 2015-06-11:
I meant when one goes into script kiddy mode and follow your readme and wants to install dump1090 on a clean raspbian you do:
wget https://github.com/mutability/mutability-repo/releases/download/v0.1.0/mutability-repo_0.1.0_armhf.deb
sudo dpkg -i mutability-repo_0.1.0_armhf.deb
sudo apt-get install dump1090-mutability

Which doesn't install rtl_test but you refer to it in the same readme :)
so suggestion was to either add it to librtl and/or add it to the readme as a sudo apt-get install rtl-sdr

closed 2015-06-26 by jester-xbmc