mutability/dump1090 #50: update librtlsdr/dump1090 to include new setting bandwidth option

jester-xbmc, 2015-06-11:
https://github.com/steve-m/librtlsdr/commit/d892279085491f9c610388da89bc701ef6ac632c

 Allow setting bandwidth for R820T

This improves SDR performence for nearby channel interference.
As a sideeffect also improves dynamic range becase ADC is not overloaded
by unwanted singles.

mutability, 2015-06-11:
Have you tried this? Given that ADS-B is a broad signal I can't see that reducing the IF bandwidth is particularly useful.

jester-xbmc, 2015-06-11:
no, not yet, building a test environment right now, even with a broad signal as ADS-B i might help to cut down on nearby interference, every uDB helps ;-)

Mictronics, 2017-03-08:
I know it's an old topic, but since I played with bandwidth settings you may interested in results. I'm running https://github.com/librtlsdr/librtlsdr v0.7.0 at the moment.

By default, bandwidth is set depending on sample frequency +/- 300KHz, so for dump1090 that will be 2.43MHz. Setting manually a lower BW will reduce noise, tracks/hour going down as well.
A higher BW, up to 6MHz, will raise noise and mean level, tracks/hour going up, but not significantly, you may receive a few aircrafts more. A BW >6MHz increases noise significantly and therefore sensitivity suffers, noticeable higher CPU load but no real benefit for reception.

So I stick with default bandwidth as it seems to be a good trade-off for low noise, good sensitivity and acceptable CPU load.

As a side effect I learned: Never use auto-gain.

closed 2019-09-14 by mutability