mutability/dump1090 #52: Preparation for systemd - daemon set PID and read config file

dschaper, 2015-06-26:
In prep for the eventual (probably?) migration to Systemd can we have the daemon create it's own pid file and instead of reading the config out as arguments from the sysv initscript, allow the daemon to directly read a config file with built in defaults? I've got dump1090 running on Arch and have most of the tarball and PKGBUILD written with the files going to their Arch counterparts, but writing the systemd  service unit file is a bit unwieldy.

mutability, 2015-06-26:
It would be nice to have. Having an external config file would probably help with running multiple copies, too.

mutability, 2015-06-26:
Also detach-and-daemonize, and write your own logfile, and reopen logs on SIGHUP. The usual daemon stuff.

dschaper, 2015-06-26:
Well, if you have the time... (Systemd actually seems to be rather nice to work with, just has kind of a steep learning curve. The socket idea seems to be rather interesting.) I saw another pull request for service files, I'll see what there is to see in that issue. Thanks again as always.

mutability, 2016-02-16:
From my playing with systemd so far you actually want the opposite of this issue!

Let systemd discover the pid.
Let systemd handle the logging to journal/syslog from stdout/stderr.

Directly reading a config file is still useful. Though you can get a similar effect by putting After/Wants on a oneshot helper service that generates an environment file.

dschaper, 2016-02-19:
I'm on the verge of abandoning Arch, and letting someone else take over the packaging for it if they would like to. I am very interested in systemD and what you've found while playing with it. I agree that `journalctl` should be the way to handle messaging, the journal logging is just so much more elegant and once you get to know how to use it you can narrow down your views pretty tightly. (Just specific units, and time periods, or just from the current boot...) 

Thanks for taking the time to respond, I know you have bigger fish to fry! (And I've learned a lot about systemD since bringing up this issue!)

dschaper, 2016-02-19:
That's an interesting take on the environmental file aspect, I had always been thinking of command line arguments/flags but I kind of can see where you're going. But wouldn't a one-shot only execute once per boot?

mutability, 2016-02-19:
One shots have the interesting property that they go inactive after they
are done. So a subsequent start or restart of the downstream service
triggers the one shot service again.

You can substitute env vars in the exec start line so this can still use
command line options.

It feels like a bit of a hack but it does seem to work and is implicitly
endorsed by the man pages which talk about using a previous service to
generate env files..

dschaper, 2016-02-19:
So if dump1090's service is restarted and has the right Wants/Wanted By, it would call the OneShot again? Interesting... Have you seen http://0pointer.de/blog/projects/systemd-for-admins-1.html (It's a long read and takes time but has some good concepts that I've been trying to wrap my head around.)

mutability, 2016-02-19:
It's the Before/After dependency, but yeah - systemd will try to start any inactive Before units before starting/restarting the downstream unit. So since the oneshot stays inactive after being run it gets run again before each start/restart

closed 2017-12-19 by dschaper