mutability/dump1090 #54: Point MLAT-positioned planes in the correct direction

mgunther68, 2015-07-04:
Currently, icons for planes where position is derived from feedback by mlat-client and/or fa-mlat-client point north (0 degrees). Perhaps it is possible to derive track info from previous positions and point the icon accordingly.
Example:
[image]

mutability, 2015-07-04:
This depends on what sort of results you are getting. If you're getting kalman filtered positions then they include velocity info; if you are getting unfiltered positions then they contain no velocity info and therefore no heading.

As deriving a track from a series of positions / measurements is basically exactly what the kalman filter is doing, the fix here is to switch to kalman positions.

(this is currently not possible with the json client, but it's on my list)

mutability, 2015-07-04:
Either way I don't think it's something that belongs in mlat-client; even if you wanted to reconstruct a track then I'd do that in dump1090 or elsewhere as it's equally applicable to any aircraft that's not sending velocity messages.

mutability, 2015-07-04:
(this is probably simplest as a little tweak to the javascript, actually - doesn't need to touch the dump1090 code at all)

mgunther68, 2015-07-04:
Thanks for looking into it!

closed 2019-09-14 by mutability