mutability/dump1090 #59: add ICAO country icon next to ICAO code in table and selected view

Dynomity, 2015-08-06:
It would be nice to see the ICAO country flag/icon in the table and selected view to attract more attention to rare flights.
I've implemented this on my own already but it'd be nice to see it incorporated in a future release.

[image]

mutability, 2015-08-06:
A pull request would be awesome..

Dynomity, 2015-08-07:
I'm new to github (not a programmer)... I'll have to read up on it and give it a try shortly.

mutability, 2015-08-09:
https://help.github.com/articles/using-pull-requests/

Short version is:

 * fork this repo
 * in your fork, create a branch and make your changes on that branch
 * create pull request from that branch

Dynomity, 2015-08-09:
Thanks,
I've done that but the code doesn't work   . If I add it to the exact place in the script.js file can you have a look at it?

It's only a few lines long and only uses the hex code so it should work but I'm no programmer. 

Dino

mutability, 2015-08-09:
If you can get me the changes in some form I can look at them, yeah.

Dynomity, 2015-08-10:
Hello Oliver,

Sure... I'd like to try to add them to a branch but I guess I'd had to
install the entire development package on a raspberry and I'm not quite
sure how to compile, etc the latest pre-release in order to test it
properly. Lots to learn!

I've included the modifications in the attached 1.14 script file for the
flag functionality.

The additions are found on lines:
46   -> declaring the lookup table variable
70-86 -> look up the flag per the ICAO code
117-118 -> append the flag to the ICAO in the main table
979-984 -> append to selected aircraft ICAO
708-732 - function to prep the image filename

Not included in this email: I added the ICAO country to the gmap.html
selected aircraft section

                                    <tr class="infoblock_body">

                                     <td colspan=1>Position: <span
id="selected_position">n/a</span></td>

                                      <td>ICAO: <span
id="selected_country">n/a</span>

</td>

   </tr>

flags are 12x20 pixels saved in the flags-tiny subdirectory.

I think I got them from http://www.iconarchive.com/category/flag-icons.html

Again sorry to the extra work. I don't have much experience with coding and
getting by mostly from seeing what others have done so the code is
problably not following proper practice. I have a fork and a branch to the
repository and will still try to go that route at some point. I'm just not
sure how to install the entire dev package in order to test it. I guess I'd
have to install fresh on a raspberry....

Please let me know what you think....

Thanks,

Dino

mutability, 2016-02-16:
This got merged.

closed 2016-02-16 by mutability