mutability/dump1090 #68: Add aircraft type (Make & Model) to the web map

skyhawk180, 2015-09-03:
It would be nice to see type of aircraft (B737 vs C206H...) either on the table or on selection, just like on FlightAware. May be there is a DB mapping for ICAO Aircraft Registry which could be leveraged here. May be there is another fork with this capabilities which someone can direct me to !. 

Thanks to Malcolm and everyone here for doing a fantastic job on home-brew ADSB trackers. I have piped the signals from this with dump978 on another dongle with great results. Also I have another pi with Oversample Turned off so ModeAC works which feeds the Primary/Production Pi. 

Proudly feeding FlightAware, PlanePlotter, FlightRadar24, LiveATC.......... and Stratux if you are interested in coupling with your iPad/ForeFlight app (https://github.com/cyoung/stratux) all with dump1090-mut/dump978.
Tailwinds !

mutability, 2015-09-03:
The current code has some support for this already; there is a static database that is populated from the VRS BasicAircraftLookup database which provides registrations and ICAO aircraft types for some aircraft; this shows up in the selected aircraft info.

It's relatively easy to add new columns to that DB or to import from different sources if you have some other source you want to use.

skyhawk180, 2015-09-03:
Thanks Oliver. Appreciate the mutability release.I will play with the database to see if I can include aircraft M&M as well.

j0nson, 2015-09-03:
I've got a fork with this enabled, I will clean it up and make a pull req this weekend.

skyhawk180, 2015-09-03:
Thanks. Let me know and I will be open to testing/updating the code as required.

j0nson, 2015-09-05:
I just added a pull request that adds this data to the gmap.  Check it out here:
https://github.com/mutability/dump1090/pull/69

mutability, 2016-02-16:
This is going to need a pull request for anything further to happen.

closed 2016-02-16 by mutability