mutability/dump1090 #76: copyright file does not mention public_html/*.js

rgeissert, 2015-10-24:
Files such as markers.js contain content released under a license that is not covered by the debian/copyright file.

dschaper, 2015-10-27:
Just to follow up on this one, here's the files that are missing and the license type if I could decipher:
public_html/markers.js   by Freepik CC BY 3.0 by Kaboldy CC BY-SA 3.0
public_html/coolclock/coolclock.js Copyright 2010, Simon Baird Released under the BSD License.
public_html/coolclock/excanvas.js Copyright 2006 Google Inc. Apache License, Version 2.0
compat/clock_gettime/* Copyright (c), MM Weiss
compat/clock_nanosleep/* Copyright © 2006 Rémi Denis-Courmont.

rgeissert, 2015-11-24:
The licence and copyright of _generic_plane_svg isn't clear to me. I started looking at the history but IIRC it comes down to a big dump of code with no specific licence.

Note that getting this cleared up is a requirement for trying to get dump1090 into Debian, and therefore to its derivatives.

mutability, 2015-11-24:
The commit that added that SVG is this one, which predates my work on dump1090: https://github.com/MalcolmRobb/dump1090/commit/5f0e295580c34da34ecef3a37f03e9a9d57485ff

I have no further information about it; I can only assume it's covered by the original BSD license

mutability, 2016-02-16:
Maintaining the copyright metadata is a low priority since the information is already present in the source; if someone wants to do the archaeology and submit a pull request, great, but otherwise this isn't going to happen any time soon.

closed 2016-02-16 by mutability