mutability/dump1090 #79: dpkg-reconfigure throws error when extra_args are already defined

dschaper, 2015-10-25:
Extra_Args are already defined as EXTRA_ARGS="--dcfilter --measure-noise". When running dpkg-reconfigure, debconf fails after 'dump1090 can log all messages' question with error 

"Invalid option "--dcfilter --measure-noise"
debconf: dialog output the above errors, giving up!

mutability, 2015-10-25:
This is a bit weird because I have installs with EXTRA_ARGS set and they are fine

dschaper, 2015-10-25:
I've never had a problem before, I'll do some more digging around, I have a few other anomalies that are coming up on a test box running Wily, like dump1090-mutability dropping stats to the logfile multiple times per second. I wonder if Ubuntu has a bug in their debconf? I'll do more research on it and see where I can reproduce the activity and where it's okay...

mutability, 2015-10-25:
You can turn on a bunch of debugging if you set DEBCONF_DEBUG=developer:

```
$ sudo bash
# DEBCONF_DEBUG=developer dpkg-reconfigure dump1090-mutability
```

dschaper, 2015-10-25:
Thanks, I'll play around with that. Learning how to package deb's now, since Jessie came out for Raspbian I'm dropping Arch except for the VPN node...

dschaper, 2015-10-25:
Okay, it has something to do with this version of dialog. 

    DEBCONF_DEBUG=developer dpkg-reconfigure dump1090-mutability

Throws an error, but readline is just fine

    DEBCONF_DEBUG=developer dpkg-reconfigure dump1090-mutability -f readline

closed 2017-12-19 by dschaper