mutability/dump1090 #82: Hanging software

M0CRT, 2015-11-11:
Hi All

Seems that the software hangs on my RPi and can only be restarted once I conduct a ctrl-c.  The web interface dies as well as any decoding.  No errors appear and I don't get any CB messages.

If the application is ended when running ok, I get the stats output...not if it's truly hung.

The hang can occur between 5m and 1h after initial startup.

I wonder if this could be in anyway attributed to an overheating of the rtl-sdr?  It's on a mast on the end of a ten meter USB hub (separate usb hub psu and running a 2A PSU on the RPi).  If it was power I would usually see the CB 1 message I believe.

The dongle is on the mast in a container but is foil wrapped...do wonder if I am experiencing an overheat...if so, I would how I can tell?

Any advice or thoughts greatly appreciated.

Nigel

mutability, 2015-11-11:
I have seen dongles occasionally just stop delivering data entirely; it might be that.

Is dump1090 still using the normal level of CPU when it hangs? Or does it spike to 100% or go to 0%?

M0CRT, 2015-11-11:
Hi

And thanks for the reply.

The CPU certainly doesn't spike and runs at around 15% when functional...about 10% extra at 25% when  hung.

Any deeply hidden logs I can dig out to see if this is a software or hardware issue?

Thanks

Nige

ledfinger, 2015-12-15:
I was experiencing hanging as well recently but when placed on the test bench it worked fine realised that the USB wireless adapter was the culprit and replaced it with a new adapter and all is working fine now.

closed 2019-09-14 by mutability