mutability/dump1090 #92: Errors were encountered while processing: dump1090-mutability

jprochazka, 2016-01-24:
For completeness I am posting the output of the build as well as the error encountered when trying to install the package. I am building and installing dump1090-mutability on a clean installation of a fully updated Debian Jessie in an amd64 environment.

The build looks good.

joe@debian:~/dump1090$ dpkg-buildpackage -b
dpkg-buildpackage: source package dump1090-mutability
dpkg-buildpackage: source version 1.15~dev
dpkg-buildpackage: source distribution UNRELEASED
dpkg-buildpackage: source changed by Oliver Jowett <oliver@mutability.co.uk>
dpkg-buildpackage: host architecture amd64
 dpkg-source --before-build dump1090
 fakeroot debian/rules clean
dh clean
   dh_testdir
   dh_auto_clean
        make -j1 clean
make[1]: Entering directory '/home/joe/dump1090'
rm -f *.o compat/clock_gettime/*.o compat/clock_nanosleep/*.o dump1090 view1090 faup1090 cprtests crctests
make[1]: Leaving directory '/home/joe/dump1090'
   dh_clean
        rm -f debian/dump1090-mutability.substvars
        rm -f debian/dump1090-mutability.*.debhelper
        rm -rf debian/dump1090-mutability/
        rm -f debian/*.debhelper.log
        rm -f debian/files
        find .  \( \( \
                \( -path .\*/.git -o -path .\*/.svn -o -path .\*/.bzr -o -path .\*/.hg -o -path .\*/CVS \) -prune -o -type f -a \
                \( -name '#*#' -o -name '.*~' -o -name '*~' -o -name DEADJOE \
                 -o -name '*.orig' -o -name '*.rej' -o -name '*.bak' \
                 -o -name '.*.orig' -o -name .*.rej -o -name '.SUMS' \
                 -o -name TAGS -o \( -path '*/.deps/*' -a -name '*.P' \) \
                \) -exec rm -f {} + \) -o \
                \( -type d -a -name autom4te.cache -prune -exec rm -rf {} + \) \)
        rm -f *-stamp
 debian/rules build
dh build
   dh_testdir
   dh_auto_configure
   debian/rules override_dh_auto_build
make[1]: Entering directory '/home/joe/dump1090'
dh_auto_build -- 'EXTRACFLAGS=-DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\"'
        make -j1 "EXTRACFLAGS=-DHTMLPATH=\\\"/usr/share/dump1090-mutability/html\\\" -DMODES_DUMP1090_VARIANT=\\\"dump1090-mutability\\\""
make[2]: Entering directory '/home/joe/dump1090'
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W `pkg-config --cflags librtlsdr` -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c dump1090.c -o dump1090.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c anet.c -o anet.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c interactive.c -o interactive.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c mode_ac.c -o mode_ac.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c mode_s.c -o mode_s.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c net_io.c -o net_io.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c crc.c -o crc.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c demod_2000.c -o demod_2000.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c demod_2400.c -o demod_2400.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c stats.c -o stats.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c cpr.c -o cpr.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c icao_filter.c -o icao_filter.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c track.c -o track.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c util.c -o util.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c convert.c -o convert.o
gcc -g -o dump1090 dump1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o crc.o demod_2000.o demod_2400.o stats.o cpr.o icao_filter.o track.o util.o convert.o -lpthread -lm -lrt `pkg-config --libs librtlsdr` -fPIE -pie -Wl,-z,relro -Wl,-z,now
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c view1090.c -o view1090.o
gcc -g -o view1090 view1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o crc.o stats.o cpr.o icao_filter.o track.o util.o -lpthread -lm -lrt -fPIE -pie -Wl,-z,relro -Wl,-z,now
make[2]: Leaving directory '/home/joe/dump1090'
make[1]: Leaving directory '/home/joe/dump1090'
   dh_auto_test
        make -j1 test
make[1]: Entering directory '/home/joe/dump1090'
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W  -c cprtests.c -o cprtests.o
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W  -g -o cprtests cpr.o cprtests.o -lm
./cprtests
testCPRGlobalAirborne[0,EVEN]: PASS
[... 218 lines cut ...]
   dh_md5sums
        (cd debian/dump1090-mutability >/dev/null ; find . -type f ! -path "./usr/share/dump1090-mutability/html/config.js" ! -path "./usr/share/dump1090-mutability/html/gmap.html" ! -path "./usr/share/dump1090-mutability/html/script.js" ! -path "./usr/share/dump1090-mutability/html/planeObject.js" ! -path "./usr/share/dump1090-mutability/html/style.css" ! -path "./usr/share/dump1090-mutability/html/markers.js" ! -path "./usr/share/dump1090-mutability/html/formatter.js" ! -path "./usr/share/dump1090-mutability/html/dbloader.js" ! -path "./etc/init.d/dump1090-mutability" ! -path "./etc/lighttpd/conf-available/89-dump1090.conf" ! -p [... cut ...]
Selecting previously unselected package dump1090-mutability.
(Reading database ... 36734 files and directories currently installed.)
Preparing to unpack dump1090-mutability_1.15~dev_amd64.deb ...
Unpacking dump1090-mutability (1.15~dev) ...
Setting up dump1090-mutability (1.15~dev) ...
dpkg: error processing package dump1090-mutability (--install):
 subprocess installed post-installation script returned error exit status 10
Processing triggers for systemd (215-17+deb8u2) ...
Errors were encountered while processing:
 dump1090-mutability

jprochazka, 2016-01-24:
Looking into things further it appears the package did install. rerunning the command sudo dpkg -i dump1090-mutability_1.15~dev_amd64.deb results in the configuration screen appearing asking for the user to run dump-1090 then where to place the log files. I specified the user joe which I am curenttly logged in as and /var/log/ and /var/log/dump1090-mutability. Both times resulted in the following output.

joe@debian:~$ sudo dpkg -i dump1090-mutability_1.15~dev_amd64.deb
(Reading database ... 36996 files and directories currently installed.)
Preparing to unpack dump1090-mutability_1.15~dev_amd64.deb ...
Unpacking dump1090-mutability (1.15~dev) over (1.15~dev) ...
Setting up dump1090-mutability (1.15~dev) ...
dpkg: error processing package dump1090-mutability (--install):
 subprocess installed post-installation script returned error exit status 10
Processing triggers for systemd (215-17+deb8u2) ...
Errors were encountered while processing:
 dump1090-mutability

mutability, 2016-01-25:
If you turn on debconf debugging it'll probably tell you something useful.

jprochazka, 2016-01-25:
ran sudo dpkg --debug=3773 -i  dump1090-mutability_1.15~dev_amd64.deb.

Everything looks good as far as the installation goes. There is A LOT of output of course so only posting the output pertaining to the error. However if you want the full output I can get that to you.

dpkg: error processing package dump1090-mutability (--install):
 subprocess installed post-installation script returned error exit status 10
D000001: ensure_diversions: same, skipping
Processing triggers for systemd (215-17+deb8u2) ...
D000002: fork/exec /var/lib/dpkg/info/systemd.postinst ( triggered /etc/init.d )
D000001: ensure_diversions: same, skipping
Errors were encountered while processing:
 dump1090-mutability

mutability, 2016-01-25:
see here: http://mdcc.cx/debian/debugging_debian_package_installations.html

Basically this will just be that I broke something in the packaging when I blind-modified it without any testing; it'll get fixed up when I have some time to test it, probably tomorrow

jprochazka, 2016-01-25:
Sounds good need any more info let me know.

mutability, 2016-01-25:
```
# DEBCONF_DEBUG=developer DEBIAN_FRONTEND=noninteractive sh -x /var/lib/dpkg/info/dump1090-mutability.postinst configure 1.15~dev
[...]
debconf (developer): starting /var/lib/dpkg/info/dump1090-mutability.postinst configure 1.15~dev
[...]
debconf (developer): <-- GET dump1090-mutability/net-fatsv-port
debconf (developer): --> 10 dump1090-mutability/net-fatsv-port doesn't exist
```

commit 4711eaf7 (2016-01-25): Remove reference to net-fatsv-port in postinst.

Fixes #92.
closed 2016-01-25 by mutability