mutability/dump1090 #93: BasicAircraftLookup.sqb is deprecated.

caiusseverus, 2016-01-26:
VRS has changed the way it supports lookup of basic aircraft data. They now use online lookups from PlanebaseNG to get information for unknown aircraft and add them to the local database.

The current version of BasicAircraftLookup.sqb will no longer be updated but remain available, so while nothing breaks, it will become more inaccurate over time.

VRS issue tracker:
https://trello.com/c/PdFDVNza

mutability, 2016-01-27:
OK; do you know of another suitable source for static data?

caiusseverus, 2016-01-27:
The only one that seems to be around now is the Planebase pre-populated sqb, but it seems you have to use their software to download it, and for now at least they seem not to be letting anyone new download it. I did have a copy, but they have updated to a new version and will have to contact them to get the new one. In any case, the full version of their software is a database front end and not suitable for automated downloading on something like a raspberry pi.

Planebase obviously have some sort of API that VRS are making use of, but I don't know under what terms they make it available.

mutability, 2016-01-27:
I did a little digging in the VRS source code, and the endpoint which VRS currently uses won't be usable as-is as it's not setting CORS headers so the javascript can't directly use it (and I don't know if Andrew intended that endpoint to be more widely used anyway).

So I'm not sure if there's actually anything to do here..

mutability, 2016-02-05:
I guess what I can do is extract the current copy and include that as static data and skip the download/extract bits in the package itself.

mutability, 2016-02-16:
108b40291848dc8a878d0cb602d6e073473975c9 includes a copy of the json derived from the current DB directly, and removes the download/update cronjob and tools from the package (the tools are still in the source tree, just not installed any more)

closed 2016-02-16 by mutability