mutability/dump1090 #94: Cross-compilation on OS X for Linux broken by PR 67

mikeage, 2016-01-27:
It seems that PR #67 broke the ability to cross-compile dump1090 on OS X for a Linux target. Specifically, I'd like to compile within OpenWrt. 

Without autoconf, I didn't see an easy way to detect this scenario, which is why I'm opening an issue, rather than a PR. Perhaps, though, replacing the existing makefile with an autotools based solution might be a good idea, though having done that for a few packages, it's never a fun task. If I have some time I might try and write up a PR to use ./configure if that seems like something you'd be interested, but I can't commit to it. 

I've submitted a PR to OpenWrt for now that simply skips this check, FYI: https://github.com/openwrt/packages/pull/2320 , but I thought you might be interested in knowing that at least one person out there is interested ;)

mutability, 2016-01-27:
I'm not a huge fan of autoconf; if there is a simpler way that needs a little support from dump1090 itself that's fine. e.g. can you just pass UNAME as a command-line var to make?

mikeage, 2016-01-27:
good point; that works. Closing...

closed 2016-01-27 by mikeage