mutability/dump1090 #96: Position errors

wnagele, 2016-02-02:
I have noticed occasional incorrect positions that are so far off that they clearly stand out. These are from frames with valid parity and I do not believe the aircrafts to send bogus frames intentionally - as these only pop up every now and then while they provide correct position information all the other times.

To help narrow the problem down I have plotted about 24h of positions received by my station in Vienna and notice a fairly clear pattern in the error data.

[image]

I am assuming that these are due to some bug in the algorithm used to extract the positions - however this is just a guess. As you can see these are a fair number of errors over the course of a 24h time window.

Anybody got an idea what might be going on there?

mutability, 2016-02-02:
Do you have a raw frame (or ideally pair of frames) that shows the problem?

mutability, 2016-02-02:
I just captured a bunch of data here locally and I do not see the same effect, so I'm definitely going to need more info to do something with this.

Are they perhaps surface positions?

mutability, 2016-02-02:
The other question here would be: where are you getting the position data from for your map, exactly?

wnagele, 2016-02-02:
These should be around the Vienna, Austria area but end up in the North Sea and Poland.

Here you go:
*8D4009C758B987323152F262F15F;
*8D4009C758B983CD6B5CF0AACF38;
*8D4009C758B987323152F262F15F;
*8D4009C758B983CF795BB6125EC1;
*8D4009C758B987323152F262F15F;
*8D4009C758B983D1255AB70A3056;
*8D4400F9901D6779BD925113B761;
*8D4400F99015600C01A94E6095E2;
*8D4400F9901DC778E79263000B40;
*8D4400F99015600C01A94E6095E2;
*8D4400F9901DD778D19265844232;
*8D4400F99015600C01A94E6095E2;
*8D4400F99033E75FED929871AE40;
*8D4400F9902343FB57AA31B51DAF;

wnagele, 2016-02-02:
Position data is pulled from port 30003 and then plotted onto a Google Map.

Here is the output from the port 30003 decoded data:
```
MSG,7,111,11111,4009C7,111111,2016/02/02,19:53:05.998,2016/02/02,19:53:05.998,,36000,,,,,,,,,,0
MSG,3,111,11111,4009C7,111111,2016/02/02,19:53:05.998,2016/02/02,19:53:05.998,,36000,,,53.70360,7.00991,,,,,,0
MSG,7,111,11111,4009C7,111111,2016/02/02,19:53:05.998,2016/02/02,19:53:05.998,,,,,,,,,,,,
MSG,7,111,11111,4009C7,111111,2016/02/02,19:53:05.998,2016/02/02,19:53:05.998,,36000,,,,,,,,,,0
MSG,7,111,11111,4009C7,111111,2016/02/02,19:53:05.999,2016/02/02,19:53:05.999,,,,,,,,,,,,
MSG,7,111,11111,4009C7,111111,2016/02/02,19:53:05.999,2016/02/02,19:53:05.999,,36000,,,,,,,,,,0
MSG,7,111,11111,4400F9,111111,2016/02/02,19:53:05.999,2016/02/02,19:53:05.999,,4750,,,,,,,,,,0
MSG,3,111,11111,4400F9,111111,2016/02/02,19:53:05.999,2016/02/02,19:53:05.999,,3150,,,54.07031,18.82978,,,,,,0
MSG,7,111,11111,4400F9,111111,2016/02/02,19:53:05.999,2016/02/02,19:53:05.999,,,,,,,,,,,,
MSG,7,111,11111,4400F9,111111,2016/02/02,19:53:05.999,2016/02/02,19:53:05.999,,3150,,,,,,,,,,0
MSG,7,111,11111,4400F9,111111,2016/02/02,19:53:05.999,2016/02/02,19:53:06.000,,,,,,,,,,,,
MSG,7,111,11111,4400F9,111111,2016/02/02,19:53:06.000,2016/02/02,19:53:06.000,,3150,,,,,,,,,,0
MSG,7,111,11111,4400F9,111111,2016/02/02,19:53:06.000,2016/02/02,19:53:06.000,,,,,,,,,,,,
MSG,7,111,11111,4400F9,111111,2016/02/02,19:53:06.000,2016/02/02,19:53:06.000,,5900,,,,,,,,,,0
```

wnagele, 2016-02-02:
No, they are airborne positions.

mutability, 2016-02-02:
The data for 4400F9 looks like it's just garbage. It is reporting a NUCp of 0 and the altitude is varying wildly (4750 - 3150 - 4900 - 3150 - 4925 - 3150 - 9350 - 5900).

mutability, 2016-02-02:
The data for 4009C7 looks more plausble, but I am pretty suspicious about the even-mode messages as it has reported exactly the same data for all 3 even messages (the 1st, 3rd, and 5th message in your sample there); that data should be changing normally if the aircraft is moving.

wnagele, 2016-02-02:
I'll do another direct capture here as these frames were filtered from an archive and the dates are not real cause I replayed them against dump1090. I think the numbers are reasonable for an aircraft approaching VIE for landing. Just assume the frames are out-of-order here and ignore the time (they are within a 15 minute timeframe).

I looked up the aircrafts and one is a Fokker and the other a Boeing. So I don't suspect a faulty transponder here.

mutability, 2016-02-02:
If I use a receiver location near Vienna and feed the messages in slowly, to force use of receiver-relative decoding from single messages rather than using message pairs for global decoding, then for 4009C7 the even and odd messages disagree on the position by a fair amount:

1: 47.587187, 15.341562
2: 47.703598, 15.133667 (20km away from 1)
3: 47.587187, 15.341562
4: 47.715637, 15.112106 (22km away from 3)
5: 47.587187, 15.341562
6: 47.725433, 15.094597 (24km away from 5)

The pairs _should_ be reporting positions that are close together (to within 0.5s travel time or so) so there's something obviously wrong there.

For 4400F9, a similar effect:

1: 48.013513, 16.484070
2: 48.070312, 16.476059 (6.3km away from 1)
3: 48.008531, 16.485337
4: 48.070312, 16.476059 (6.9km away from 3)
5: 48.008019, 16.485478
6: 48.070312, 16.476059 (6.9km away from 5)
7: 47.859704, 16.489070
8: 47.972672, 16.491646 (12.6km away from 7)

I suspect if you look for outlier positions (say, anything >400km away from your receiver) and record the aircraft addresses, you'll find it is a small number of aircraft that are causing this - bad transponder data, most likely.

Note that you can mask out most of these bad positions by configuring an appropriate receiver location and max range.

mutability, 2016-02-02:
Yeah, the timing is important, you need to be careful when replaying to get approximately the same (realtime) spacing of messages as dump1090 makes a number of decisions about position decoding based on the interval between messages.

mutability, 2016-02-02:
In particular, if you replay much faster than realtime then dump1090 has two problems:

1) it may believe that an even/odd pair that were transmitted a long time apart are actually close enough to be used as a pair for global position decoding. This would lead to pretty much exactly what you see on the map: positions being assigned to the wrong cell (the cells are around 360NM wide).
2) it confuses the speed checks as aircraft seem to be moving much faster than they really are, so many valid positions can be discarded.

Did you construct the map from a replay, or in realtime?

wnagele, 2016-02-02:
Yes - I captured 24h on port 30003 and then ran them against dump1090 to get the SBS1 output. Didn't have a clue that positions span messages and this could cause issues. Might be a good idea to document that the input ports should be used for piping in data from another running dump1090 instance to avoid these issues? I always thought of this as a nice debugging tool as well but it obviously can cause seriously wrong output data. Thoughts?

mutability, 2016-02-02:
It would be nice to use the packet timestamps to handle the timing; I don't know what else can be done here.

wnagele, 2016-02-02:
The normal packet payload here doesn't have timestamps though - does it? In my case I can adjust the replay as I am keeping the original timestamp along with the payload. Takes much longer to get the results this way but at least they are correct.

mutability, 2016-02-02:
The AVR format doesn't have timestamps, no, but Beast or AVR with mlat timestamps does. They are 12MHz clock timestamps but that would be enough to drive the calculation of time intervals.

wnagele, 2016-02-03:
Thanks for the quick help with this. Should give enough information if anybody else encounters such an issue in the future. Closing.

closed 2016-02-03 by wnagele