mutability/dump1090 #97: Max range not working

wesleygoatley, 2016-02-04:
I can't seem to make the max-range argument respond as it seems it should - running dump1090 --max-range 50 --net --interactive still clearly gives me a lot of contacts 100km+ away.  Is this a known issue, or something to do with it running net/interactive?

mutability, 2016-02-04:
Looks like there is a bug where synthetic mlat positions are always accepted regardless of range. I can't reproduce this with off-the-air messages though.

closed 2016-02-05 by mutability

wesleygoatley, 2016-02-05:
> Closed #97 <https://github.com/mutability/dump1090/issues/97> via e101351

Thanks Oliver - just checked with your new commit, and I'm getting the same
issue - screenshot attached (I'm in Leuven, scale marker visible at the
bottom)

[image: Inline image 1]

mutability, 2016-02-05:
Can you resend the image directly to me or post a link or attach it via the github web interface, it looks like github's email-to-issue-comment gateway discards attachments

commit e101351b (2016-02-05): Enforce max-range on mlat results (but still skip the speed check)

Fixes #97