mutability/dump1090 #100: decoded output has 111s for flight_id, session_id, aircraft_id

jeremybmerrill, 2016-02-10:
Hi Oliver,

I'm running your fork of dump1090 on raspbian/a raspberry pi. The port 30003 feed of decoded messages always has 1s for flight_id, session_id and aircraft_id. I'm not sure why this is, but I think it's a bug.

The web interface works fine, including flight_id. (I'm not 100% sure what the other two should look like).

Here's an example:

```
MSG,7,111,11111,AD4B03,111111,2016/02/10,09:02:04.702,2016/02/10,09:02:04.733,,1950,,,,,,,,,,0
MSG,7,111,11111,A7F27A,111111,2016/02/10,09:02:04.726,2016/02/10,09:02:04.735,,4250,,,,,,,,,,0
MSG,7,111,11111,A7F27A,111111,2016/02/10,09:02:04.743,2016/02/10,09:02:04.787,,4250,,,,,,,,,,0
MSG,7,111,11111,A43FBA,111111,2016/02/10,09:02:04.755,2016/02/10,09:02:04.788,,10625,,,,,,,,,,0
MSG,7,111,11111,A7F27A,111111,2016/02/10,09:02:04.768,2016/02/10,09:02:04.789,,4250,,,,,,,,,,0
MSG,8,111,11111,A7F27A,111111,2016/02/10,09:02:04.770,2016/02/10,09:02:04.789,,,,,,,,,,,,0
MSG,7,111,11111,A7F27A,111111,2016/02/10,09:02:04.774,2016/02/10,09:02:04.789,,4250,,,,,,,,,,0
MSG,7,111,11111,A43FBA,111111,2016/02/10,09:02:04.785,2016/02/10,09:02:04.841,,10625,,,,,,,,,,0
MSG,7,111,11111,A7F27A,111111,2016/02/10,09:02:04.803,2016/02/10,09:02:04.842,,4250,,,,,,,,,,0
MSG,7,111,11111,A7F27A,111111,2016/02/10,09:02:04.811,2016/02/10,09:02:04.842,,4250,,,,,,,,,,0
MSG,7,111,11111,A7F27A,111111,2016/02/10,09:02:04.814,2016/02/10,09:02:04.843,,4250,,,,,,,,,,0
MSG,8,111,11111,A43FBA,111111,2016/02/10,09:02:04.838,2016/02/10,09:02:04.844,,,,,,,,,,,,0
MSG,7,111,11111,A7F27A,111111,2016/02/10,09:02:04.875,2016/02/10,09:02:04.898,,4250,,,,,,,,,,0
MSG,7,111,11111,A7F27A,111111,2016/02/10,09:02:04.930,2016/02/10,09:02:04.952,,4250,,,,,,,,,,0
MSG,7,111,11111,A43FBA,111111,2016/02/10,09:02:04.932,2016/02/10,09:02:04.952,,10625,,,,,,,,,,0
MSG,7,111,11111,A7F27A,111111,2016/02/10,09:02:05.001,2016/02/10,09:02:05.008,,4250,,,,,,,,,,0
MSG,7,111,11111,A7F27A,111111,2016/02/10,09:02:05.020,2016/02/10,09:02:05.060,,4250,,,,,,,,,,0
MSG,8,111,11111,A7F27A,111111,2016/02/10,09:02:05.031,2016/02/10,09:02:05.061,,,,,,,,,,,,0
MSG,8,111,11111,A43FBA,111111,2016/02/10,09:02:05.084,2016/02/10,09:02:05.115,,,,,,,,,,,,0
MSG,7,111,11111,A43FBA,111111,2016/02/10,09:02:05.085,2016/02/10,09:02:05.115,,10650,,,,,,,,,,0
MSG,5,111,11111,A0EF27,111111,2016/02/10,09:02:05.095,2016/02/10,09:02:05.116,,12350,,,,,,,0,,0,0
MSG,5,111,11111,A0EF27,111111,2016/02/10,09:02:05.103,2016/02/10,09:02:05.116,,12350,,,,,,,0,,0,0
```

Thanks for your help!

mutability, 2016-02-10:
They're dummy values. dump1090 has no concept of session IDs etc, they are a basestation-only thing.

closed 2016-02-10 by mutability

jeremybmerrill, 2016-02-10:
Thanks for your quick response. What about flight id? The web interface knows about flight numbers, so I'm assuming the port 30003 feed ought to.

mutability, 2016-02-10:
The transmitted callsign/ident is fed in column 11 (if I counted right - the first column after the timestamp) where available; that's the closest you'll get to a flight ID.

jeremybmerrill, 2016-02-10:
Ah, that's what I'm looking for then. Thanks for your help.