mutability/dump1090 #105: Surface position inaccuracy

hbokh, 2016-02-28:
The actual position of my receiver is 53.2151388N 6.5709272E
https://goo.gl/maps/dx5wd7bUVum

but somehow `dump1090-mutability` always puts it on 
https://goo.gl/maps/Ptc5SKC67cD2

which is like ~ 800m / half a mile up north.
It makes no difference in using more accuracy in the options, e.g. `--lat 53.2151388 --lon 6.5709272` is giving the same position as `--lat 53.215 --lon 6.571`.
Is this something I'm doing wrong or something else.

Cheers, Henk

PS Thanks for a great piece of software though!

mutability, 2016-02-28:
Set --json-location-accuracy 2 if you want to report the exact receiver position; it is rounded for display by default for privacy reasons

closed 2016-02-28 by mutability

gonzalu, 2018-11-14:
Unable to set this variable in /etc/default/dump1090-mutability

I have tried using 
>JSON_LOCATION_ACCURACY="exact"

daemon still shows using option 1 (approximate)

Also, default should be 0 if not defined in config as per init.d script.

Thanks for your help