mutability/dump1090 #108: Please add SDRPlay to tree

JonFear, 2016-03-25:
SDRPlay have released a version of dump1090 at https://github.com/SDRplay/dump1090 for the SDR Play. Can you please merge their version with yours. Only reason is I want one version and quite simply yours is better than all other versions...

mutability, 2016-03-25:
I'm happy to look at a suitable pull request.

The upstream repository has discarded the git history so I cannot easily
look at the changes they have made versus a common ancestor.

JonFear, 2016-03-25:
Sorry Oliver, does this mean I need to do anything for you to be able to do this. Being a bit thick this morning! Jon (360radar & MLAT Radar).

mutability, 2016-03-25:
You (or someone else) needs to work out the set of changes that you want merged and send them to me as a patch or as a pull request on github.

jester-xbmc, 2016-03-28:
I think this commit has the basics, I haven't checked it (don't own an sdrplay) but should be pretty straightforward for Jon to whip up a pull request.
 
https://github.com/itemir/dump1090_sdrplus/commit/e33e8ea6e782f44324aa0ce6311980018ee4a0bc

rxseger, 2016-06-07:
👍 or all of [dump1090_sdrplus](https://github.com/itemir/dump1090_sdrplus), which would not only add SDRplay support but also HackRF One and Airspy

mutability, 2016-06-07:
This needs a pull request.

closed 2016-06-07 by mutability

rxseger, 2016-06-09:
If anyone is interested, I've been looking into porting over the added SDR compatibility from dump1090_sdrplus to dump1090-mutability (starting with the HackRF since that's the only device I have atm, but others should be straightforward once one is ported), but there have been many changes since the fork, it is not a clean patch to apply. 

https://gist.github.com/rxseger/71f62fcdd764fee0132ac96f725d390e

Some of the changes are clear how they port over, but the callback (rtlsdrCallback / hackrfCallback) is completely different, would likely need to be rewritten.

JonFear, 2016-06-09:
I certainly would be interested being completely code illiterate! I no longer wish for SDR Play compatibility as I went down the Airspy Mini route, that is where my interest lies now. Someone already has a working dump1090 but based on the original: https://github.com/itemir/dump1090_sdrplus. In addition, I suspect that the missing elements are released in the airspy git: https://github.com/airspy/host however being of no use with code, this is well beyond my capabilities...I can test with my Airspy Mini at any time...Thanks, Jon

rxseger, 2016-07-03:
@mutability would you be interested in having dump1090 use some kind of abstraction layer to support various SDR devices, instead of directly interfacing to librtlsdr? [SoapySDR](https://github.com/pothosware/SoapySDR/wiki) seems like a good choice:

[image]