mutability/dump1090 #113: Hardware Abstraction Layer

shepherd007, 2016-04-02:
Today I tried to compile dump1090 natively for windows (I was surprised it worked - some specific functions such as nonsleep were stubbed without implementing anything). I am wondering; wouldn't be beneficial for future porting to add some kind of HAL? Let's say - we have a few pthread functions that could be easily replaced by native windows calls - we can add common header with API declarations and implementation would be specific for the OS. The same thing could apply to the networking.
I am rather thinking of covering low level system functions with HAL API than using #ifdef in the production code.

mutability, 2016-04-02:
No fundamental objection so long as it doesn't mangle the existing code too much. There is something like it already going on for OSX / OpenBSD support (clock_gettime et al). If you want to build implementations of whatever Windows etc is missing, that's great. Do you have a list of what's needed?

shepherd007, 2016-04-03:
Yeah, I've got some ideas - and today I enabled networking on the windows (I really like the improvements on HTML GUI!). I'd like to push some of updates to windows_development branch - however it seems I have no push access. Could you please add me as a contributor?

mutability, 2016-04-03:
I suggest you fork the repository and work in your copy, then use pull requests.

shepherd007, 2016-04-05:
Here is a list of thing I am seeing as a candidates for abstraction layer (for now):
- signaling - it looks that function such like initSignaling() would be enough both for posix and windows
- console operations - currently only clr() function is needed
- net_io.c is using directly close() function on the socket - it is also a good candidate to be reworked/moved to another file

mutability, 2016-04-06:
Yeah, from looking through your pull request that list seems reasonable.

closed 2019-09-14 by mutability