mutability/dump1090 #115: Improved aircraft categorisation

rgeissert, 2016-04-05:
The current map markers are based on the FAA categories but few aircraft announce theirs. It would be great to improve this situation so that adding new icons is actually worth the effort.

rgeissert, 2016-04-05:
That said, I've been building a mapping between the ICAO type and the FAA categories of surrounding aircraft.

Here's the sample:
```json
{"A30B": "A5", "A310": "A5", "A319": "A3", "A320": "A3", "A321": "A3", "A332": "A5", "A333": "A5", "A388": "A5", "B738": "A3", "B763": "A5", "BE30": "A1", "C25C": "A2", "C680": "A2", "CL30": "A2", "CL35": "A2", "CL60": "A2", "E170": "A3", "E190": "A3", "F2TH": "A2", "F900": "A2", "FA7X": "A2", "GL5T": "A3", "GLEX": "A3", "GLF5": "A3", "GLF6": "A3"}
```

I've a script that waits for aircraft that announce their category and then looks up their registration to obtain the ICAO type.

What do you think?

mutability, 2016-04-05:
Setting the icon based on a combination of the broadcast aircraft category and aircraft metadata loaded from the static DB sounds fine to me.

jthawkes, 2016-04-06:
I get a lot of Helos in NYC. Tourist, Corporate Transport, Airport shuttle, DHL, Police, EMS, Military and visitors. I noticed a few are not categorized(iconned) correctly. I am happy to help out updating a list.

danaronoff, 2016-04-06:
@jthawkes I also live in NYC and I am in the process of categorizing and filtering aircraft for my project. Was there any pattern to what you noticed wasn't coded correctly? I am interested in identifying seaplanes and helicopters only.

jthawkes, 2016-04-07:
I suppose I should have asked for more details on how it is done. 
Should it work for MLAT aircraft?
I have seen a Bell 407 that had an airplane icon. I was not working from home today so could not check. I am home tomorrow so will try to catch what I can. 

I just noticed this mlat aircraft. 
http://www.airframes.org/reg/n350lh

If mlat won't work then it is not an issue.

rgeissert, 2016-04-07:
I've just pushed the script into the _improved-categorisation_ branch of my fork: https://github.com/rgeissert/dump1090/tree/improved-categorisation

As of the time of writing, my icaomaping._json is:

```json
{"A30B": "A5", "A310": "A5", "A319": "A3", "A320": "A3", "A321": "A3", "A332": "A5", "A333": "A5", "A388": "A5", "A3ST": "A4", "B738": "A3", "B748": "A5", "B763": "A5", "B789": "A5", "BE30": "A1", "BE9L": "A1", "C25C": "A2", "C680": "A2", "CL30": "A2", "CL35": "A2", "CL60": "A2", "E170": "A3", "E190": "A3", "E3CF": "A3", "F2TH": "A2", "F900": "A2", "FA7X": "A2", "GL5T": "A3", "GLEX": "A3", "GLF4": "A3", "GLF5": "A3", "GLF6": "A3"}
```

jthawkes, 2016-05-26:
What about an icon for ground vehicles. I have 3 large airports nearby, JFK, EWR and LGA and when the conditions are right, I see 5-50 ground vehicles. It would be nice to be able to tell the difference between them and aircraft on the ground.

rgeissert, 2016-05-26:
I started hacking up some icons for that as well, but I haven't centred them yet. They are all from Freepik, just like some of the existing icons.

[ground-transport.zip](https://github.com/mutability/dump1090/files/283886/ground-transport.zip)

mutability, 2017-02-09:
FWIW this is mostly superseded by putting type information in the json db (needs a backport from dump1090-fa)

closed 2019-09-14 by mutability