mutability/dump1090 #129: Tuning gain control and improving reception

vielmetti, 2016-07-02:
Is there a handy rule of thumb for when to use automatic gain control vs. manual gain control?

Before AGC, a typical RSSI for a plane nearly overhead was -15 dB; now, I'm seeing -5 dB for similar routes.

I turned on AGC with `--enable-agc` and the RSSI readings "improved" (got larger), but I can't readily make a distinction in how well airplanes are being tracked. I do know that there's room for improvement, in that I have an antenna in an attic and have about 45 nm (85 km) of effective range. What I don't know is whether I can get better range by tweaking `dump1090` software settings, or whether a next step is a better antenna or a filter or amplifier or etc.

A dump of stats:

```
Statistics: Sat Jul  2 18:11:28 2016 EDT - Sat Jul  2 18:12:37 2016 EDT
Local receiver:
  165150720 samples processed
  0 samples dropped
  32 Mode A/C messages received
  1862788 Mode-S message preambles received
    1191550 with bad message format or invalid CRC
    667793 with unrecognized ICAO address
    3445 accepted with correct CRC
  -24.8 dBFS noise power
  -8.7 dBFS mean signal power
  -2.1 dBFS peak signal power
  58 messages with signal power above -3dBFS
```

(That seems like a lot of "invalid CRC" messages, hence the suspicion that I can do better!)

keptenkurk, 2016-07-03:
@vielmetti There is quite some discussion on optimizing the gain setting to be found [here at FlightAware](http://discussions.flightaware.com/ads-b-flight-tracking-f21/gain-adjustment-t37172.html).
It seems to boil down into using an optimal fixed gain setting found by trial & error. Doing "sudo dpkg-reconfigure dump1090-mutability", change gain and watch the effect on "Messages/sec" in dump1090's web view.
A rule of thumb would need to take more factors into account like preamp yes/no and vincinity of GSM cell towers...
/paul

vielmetti, 2016-07-05:
Thanks @keptenkurk . My particular situation is preamp: no and GSM towers: plenty. I did a spectrum survey (using `rtl_power` and `CubicSDR`) looking for activity in the cell phone bands and found plenty of it, no surprise since we live near a stadium and there's perpetually not enough cell coverage during games.

Based on the FlightAware and other suggestions I'm looking at a band pass filter as the next $20 or $30 to spend, in the hopes that reduces the noise.

jthawkes, 2016-07-05:
I would agree that the bad messages(/invalid CRC) looks a little high.

I have about a 10% acceptance rate(of preambles received). 
You appear to have a 0.2% acceptance rate. Quite a difference.
I am using an FA Dongle with DPD Antenna(Slightly better than the FA dual band antenna).
I use a slightly better cavity filter and am in NYC, a really busy area for aircraft.

I left out a filter of one setup a few weeks ago. Went from 150-200 aircraft to 2.
This is with the FA dongle gain turned down to 38-40db.

Statistics: Tue Jul  5 18:29:04 2016 EDT - Tue Jul  5 19:29:04 2016 EDT
Local receiver:
  8640528384 samples processed
  0 samples dropped
  271563 Mode A/C messages received
  41748081 Mode-S message preambles received
    23118291 with bad message format or invalid CRC
    14236965 with unrecognized ICAO address
    4164861 accepted with correct CRC
    227964 accepted with 1-bit error repaired
  -16.6 dBFS noise power
  -6.7 dBFS mean signal power
  -0.5 dBFS peak signal power
  652873 messages with signal power above -3dBFS

closed 2019-09-14 by mutability