mutability/dump1090 #130: Oops! Something went wrong. This page didn't load Google Maps correctly. See the JavaScript console for technical details.

vielmetti, 2016-07-03:
I have dump1090 v1.14 running on a Pi, and am trying to get the Pi to display the map from its web server.

When I set up the connection I get this error:

Oops! Something went wrong.
This page didn't load Google Maps correctly. See the JavaScript console for technical details.

I know that things are working because the correct map flashes on the screen briefly.

The relevant error in the log files seems to be

js?sensor=false&libraries=geometry:32 Google Maps API error: MissingKeyMapError https://developers.google.com/maps/documentation/javascript/error-messages#missing-key-map-error

and the text that comes up when I search for this is

https://www.latecnosfera.com/2016/06/google-maps-api-error-missing-keymap-error-solved.html

> If you are using the Google Maps API on localhost or your domain was not active prior to June 22nd, 2016, it will require a key going forward. To fix this problem, please see the Google Maps APIs documentation to get a key and add it to your application: https://developers.google.com/maps/documentation/javascript/get-api-key

mutability, 2016-07-03:
Yeah, you will either need to provide an API key and include it in your local copy, or you can try the OpenLayers-based branch here: https://github.com/mutability/dump1090/tree/ol3

closed 2016-07-03 by mutability

vielmetti, 2016-07-03:
Thanks! That was enough of a clue to get it working. I patched `public_html/gmap.html` (at line 8) to include the API key, and all is well for now.

vielmetti, 2016-07-06:
Also helpful in this regard is https://www.latecnosfera.com/2016/06/google-maps-api-error-missing-keymap-error-solved.html which walks  you through getting an API key from Google.

mutability, 2016-07-06:
The OpenLayers changes are in master now.