mutability/dump1090 #131: Open Layers loading problems

suicider, 2016-07-06:
In the latest version with the chance from Google Maps to OSM the Open Layers files are not loading (endless spinning sign). I have the following errors in the Chrome JavaScript Console:

> ol3-layerswitcher.js:9 Uncaught ReferenceError: ol is not defined
> script.js:6 Uncaught ReferenceError: ol is not defined(anonymous function) @ script.js:6
> script.js:826 Uncaught TypeError: Cannot read property 'length' of undefined

I also tried to add a Mapzen API key, but cannot change the default map. Any Ideas?

mutability, 2016-07-06:
It works for me on chromium 51.0.2704.79; what's the full content of the console leading up to those errors?

suicider, 2016-07-06:
This is the full content:

> Failed to load resource: net::ERR_CONTENT_LENGTH_MISMATCH
> http://10.99.15.168:8080/jquery/jquery-ui-1.11.4.min.js Failed to load resource: net::ERR_CONTENT_LENGTH_MISMATCH
> http://10.99.15.168:8080/ol3/ol-3.17.1.js Failed to load resource: net::ERR_CONTENT_LENGTH_MISMATCH
> ol3-layerswitcher.js:9 Uncaught ReferenceError: ol is not defined
> script.js:6 Uncaught ReferenceError: ol is not defined(anonymous function) @ script.js:6
> script.js:826 Uncaught TypeError: Cannot read property 'length' of undefined

it's running on a Raspberry Pi in an internal network ("Freifunk") but with Internet access.

mutability, 2016-07-06:
Did you rebuild dump1090? The "content length mismatch" indicates that you're using the internal webserver and it's not patched to handle larger file sizes. I'd recommend using an external webserver anyway.

mutability, 2016-07-06:
specifically, you need this commit: https://github.com/mutability/dump1090/commit/da437b32958d5d0c275d97a8e4f2bdb2bb404fe9

suicider, 2016-07-06:
Yes, I'm using the internal webserver. Ok, thanks, I'll set up an external server but the internal was good enough until now.

vielmetti, 2016-07-21:
the internal web server is now disabled, unless you set a compile-time flag. I'd welcome suggestions for an external web server on MacOS - the homebrew version of `lighttpd` looks like it's a few revs behind of the one on the Raspberry Pi, so the provided instructions don't work out of the box. (I ended up setting the flag and rebuilding for now.)

mutability, 2016-07-24:
Pretty much anything which can serve static files should work fine, take your pick.. Even an old lighttpd should work OK if you can find the right config.

mutability, 2016-07-24:
(for the record, the workaround I tried for the internal webserver, above, isn't sufficient in many cases, which is why the internal webserver is now disabled by default - see #132 for details)

closed 2016-07-24 by mutability