mutability/dump1090 #132: Problem APIKEY for the New Version

Sondermannolaf, 2016-07-06:
Hi Guys,
I asked a problem with the config in the maps are the APIKEYS . I besort me the matching keys .
But the website is only load on the Dump1090 works and sends data .
The config looks like this:

BingsMapsAPIKey = ASDTF VALUE FTR ...... ;

MapzenAPIKey = vector - tiles- xxxxxxxx

I guess it can not connect to the Maps providers .

The dump1090 running under Ubuntu 14.04 LTS .
Best thanks in advance.

mutability, 2016-07-06:
Did you quote the values? Like this:

MapzenAPIKey = 'vector-tiles-xxxxx';

Sondermannolaf, 2016-07-06:
Hello,
 this is my config.js, MapzenAPIKey = 'vector-tiles-xxxxx'; write is ok. Do not work for the 127.0.0.8080 and load to load for the Spimmy.

// --------------------------------------------------------
//
// This file is to configure the configurable settings.
// Load this file before script.js file at gmap.html.
//
// --------------------------------------------------------

// -- Title Settings --------------------------------------
// Show number of aircraft and/or messages per second in the page title
PlaneCountInTitle = true;
MessageRateInTitle = false;

// -- Output Settings -------------------------------------
// Show metric values
// The Metric setting controls whether metric (m, km, km/h) or
// imperial (ft, NM, knots) units are used in the plane table
// and in the detailed plane info. If ShowOtherUnits is true,
// then the other unit will also be shown in the detailed plane
// info.
Metric = false;
ShowOtherUnits = true;

// -- Map settings ----------------------------------------
// These settings are overridden by any position information
// provided by dump1090 itself. All positions are in decimal
// degrees.

// Default center of the map.
DefaultCenterLat = 51.4;
DefaultCenterLon = 7.5;
// The google maps zoom level, 0 - 16, lower is further out
DefaultZoomLvl   = 7;

// Center marker. If dump1090 provides a receiver location,
// that location is used and these settings are ignored.

SiteShow    = false;           // true to show a center marker
SiteLat     = 45.0;            // position of the marker
SiteLon     = 9.0;
SiteName    = "My Radar Site"; // tooltip of the marker

// -- Marker settings -------------------------------------

// These settings control the coloring of aircraft by altitude.
// All color values are given as Hue (0-359) / Saturation (0-100) / Lightness (0-100)
ColorByAlt = {
        // HSL for planes with unknown altitude:
        unknown : { h: 0,   s: 0,   l: 40 },

        // HSL for planes that are on the ground:
        ground  : { h: 120, s: 100, l: 30 },

        air : {
                // These define altitude-to-hue mappings
                // at particular altitudes; the hue
                // for intermediate altitudes that lie
                // between the provided altitudes is linearly
                // interpolated.
                //
                // Mappings must be provided in increasing
                // order of altitude.
                //
                // Altitudes below the first entry use the
                // hue of the first entry; altitudes above
                // the last entry use the hue of the last
                // entry.
                h: [ { alt: 2000,  val: 20 },    // orange
                     { alt: 10000, val: 140 },   // light green
                     { alt: 40000, val: 300 } ], // magenta
                s: 85,
                l: 50,
        },

        // Changes added to the color of the currently selected plane
        selected : { h: 0, s: -10, l: +20 },

        // Changes added to the color of planes that have stale position info
        stale :    { h: 0, s: -10, l: +30 },

        // Changes added to the color of planes that have positions from mlat
        mlat :     { h: 0, s: -10, l: -10 }
};

// For a monochrome display try this:
// ColorByAlt = {
//         unknown :  { h: 0, s: 0, l: 40 },
//         ground  :  { h: 0, s: 0, l: 30 },
//         air :      { h: [ { alt: 0, val: 0 } ], s: 0, l: 50 },
//         selected : { h: 0, s: 0, l: +30 },
//         stale :    { h: 0, s: 0, l: +30 },
//         mlat :     { h: 0, s: 0, l: -10 }
// };

// Outline color for aircraft icons with an ADS-B position
OutlineADSBColor = '#000000';

// Outline color for aircraft icons with a mlat position
OutlineMlatColor = '#4040FF';

SiteCircles = true; // true to show circles (only shown if the center marker is shown)
// In nautical miles or km (depending settings value 'Metric')
SiteCirclesDistances = new Array(100,150,200);

// Show the clocks at the top of the righthand pane? You can disable the clocks if you want here
ShowClocks = true;

// Controls page title, righthand pane when nothing is selected
PageName = "DUMP1090";

// Show country flags by ICAO addresses?
ShowFlags = true;

// Path to country flags (can be a relative or absolute URL; include a trailing /)
FlagPath = "flags-tiny/";

// Set to true to enable the ChartBundle base layers (US coverage only)
ChartBundleLayers = true;

// Provide a Bing Maps API key here to enable the Bing imagery layer.
// You can obtain a free key (with usage limits) at
// https://www.bingmapsportal.com/ (you need a "basic key")
BingMapsAPIKey = null;

// Provide a Mapzen API key here to enable the Mapzen vector tile layer.
// You can obtain a free key at https://mapzen.com/developers/
// (you need a "vector tiles" key)
MapzenAPIKey = 'vector-tiles-JChdXXX';

This is my config.....

mutability, 2016-07-06:
a) I recommend you use a separate external webserver, not the internal webserver;
b) if you are using the internal webserver, make sure you rebuilt/restarted dump1090; you need this commit or larger files such as the jquery/openlayers javascript will not be served correctly by the internal webserver: https://github.com/mutability/dump1090/commit/da437b32958d5d0c275d97a8e4f2bdb2bb404fe9

anish, 2016-07-10:
@mutability I'm using a newly built dump1090 (v1.15-dev-220-g68ef9ce) with the above mentioned commit, cannot get any of the map stuff to work after multiple service restarts. Stuck on the loading spinner.

mutability, 2016-07-10:
Does it work OK with an external webserver?

mutability, 2016-07-10:
I assume you're both running on Pis.
The Pi has different default TCP buffer settings compared to my development PC, which is why I didn't see this.

There is no effective workaround. Use an external webserver. I'll probably disable the internal webserver entirely shortly.

closed 2016-07-10 by mutability

anish, 2016-07-10:
@mutability Will try that, however I'm not on debian. Is there a README on how to switch to an external server ?

mutability, 2016-07-10:
The short version is:

 * specify a --write-json argument that points to a temporary directory somewhere (you may want to put it on tmpfs)
 * make the contents of public_html available via your webserver at some path /foo
 * make the contents of the json directory available via your webserver at /foo/data

anish, 2016-07-10:
thanks @mutability !

anish, 2016-07-10:
@mutability Works just fine. I'll be updating the Archlinux package to default to this soon