mutability/dump1090 #134: bingmaps key

imesg, 2016-07-24:
When I edit /usr/share/dump1090-mutability/html/config.js BingMapsAPIKey = null; and change it to my key the web page only loads the clocks and table headings and nothing else i.e. no maps or table data. If changed to 'null' page displays normally.

mutability, 2016-07-24:
I can't reproduce this. Perhaps you introduced a syntax error? The config should look like this (with your own key obviously):

   BingMapsAPIKey = 'At.......or_f';

imesg, 2016-07-24:
You are correct. Entries require quotes.

imesg, 2016-07-24:
The maps from bing and mapzen are rudimentary. Bing is aereal photo and mapzen is coastline only. Anyway to get other maps?

By the way thanks for your work on this program.

mutability, 2016-07-24:
Mapzen is more of a proof of concept rather than anything more.. If you're willing to play with the styling you can control the rendering further.

I just added the Bing roads layer which might be more useful than the aerial layer.

closed 2016-07-24 by mutability