mutability/dump1090 #136: New package version of dump1090 for RPI

takenek, 2016-07-29:
Hello
Can You give new version of dump1090 for RPI ?
I use Your repository: 
deb http://repo.mutability.co.uk/raspbian wheezy rpi
But there is dump1090 from 19 February 2015.

Best Regards

vielmetti, 2016-08-04:
I was able to follow the "building from source" directions to build a new version of dump1090 for the RPi, as described at the bottom of the README page

https://github.com/mutability/dump1090/blob/master/README.md

You may need to install compilers and other build utilities as well, depending on what all else you have already installed on the Pi as a build environment. The compile was relatively quick even on a Pi 2.

anish, 2016-08-04:
You can also take a look at this file that compiles and packages dump1090 https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=dump1090-fa-git

takenek, 2016-08-18:
Hello
I know i can use a source from github i only report the version in official portage is amazing out of date...
 19 February 2015. So if You can please update.

Best Regards

kevinelliott, 2016-09-12:
I agree, I'd like to see this package updated as well. The package includes things like an init.d script for letting the system start/stop/restart/etc, as well as installs things into standard locations. It also allows for easy integration with FlightAware and other aspects.

I vote for an update to latest release!

mutability, 2016-09-12:
I have no plans to do a "proper" release in the near future, but I could perhaps set up snapshot releases of some kind if the thing here is "I want a binary package" not "I want a proper release"

kevinelliott, 2016-09-13:
Does that just mean you'll update the current Raspbian package to latest HEAD? If so, that would be great 👍

kevinelliott, 2016-09-19:
Just realized it's easy to build your own package from latest head.

1. Git clone from github (or pull if you already have it cloned)

`git clone https://github.com/mutability/dump1090.git`

or

`git pull` from the directory

2. Build the package

`dpkg-buildpackage -b`

3. Install/update the package

`sudo dpkg -i ../dump1090-mutability_1.15~dev_armhf.deb`

4. Restart the service

`sudo /etc/init.d/dump-mutability restart`

5. Voila, all done!

closed 2019-09-14 by mutability