mutability/dump1090 #137: Sending large files (ol-3.1.7.1.js) over HTTP sometimes blocks

daviessm, 2016-08-04:
I fixed this locally, in net_io.c something like this should work:

        // Send header and content.
        bytes_sent_total = 0;
        while (bytes_sent_total < hdrlen) {
    #ifndef _WIN32
            bytes_sent_this = write(c->fd, bytes_sent_total+hdr, hdrlen-bytes_sent_total);
    #else
            bytes_sent_this = send(c->fd, bytes_sent_total+hdr, hdrlen-bytes_sent_total, 0);
    #endif
            if (bytes_sent_this == -1 && errno != EWOULDBLOCK && errno != EINTR) {
                printf("Got error %d\n", errno);
                return(1);
            } else if (bytes_sent_this > 0) {
                bytes_sent_total += bytes_sent_this;
            }
            if (Modes.debug & MODES_DEBUG_NET) {
                printf("Sent data lengths:%d (%d of %d)\n", bytes_sent_this, bytes_sent_total, hdrlen);
            }
        }

        bytes_sent_total = 0;
        while (bytes_sent_total < clen) {
    #ifndef _WIN32
            bytes_sent_this = write(c->fd, bytes_sent_total+content, clen-bytes_sent_total);
    #else
            bytes_sent_this = send(c->fd, bytes_sent_total+content, clen-bytes_sent_total, 0);
    #endif
            if (bytes_sent_this == -1 && errno != EWOULDBLOCK && errno != EINTR) {
                printf("Got error %d\n", errno);
                free(content);
                return 1;
            } else if (bytes_sent_this > 0) {
                bytes_sent_total += bytes_sent_this;
            }
            if (Modes.debug & MODES_DEBUG_NET) {
                printf("Sent data lengths:%d (%d of %d)\n", bytes_sent_this, bytes_sent_total, clen);
            }
        }
        free(content); 
        Modes.stats_current.http_requests++;

I'm not sure about the return values for the win32 code. Someone should check it over.

mutability, 2016-08-04:
Yeah, but this blocks dump1090 (worse, in a busy loop..) until the write completes, which is no good; meanwhile you are dropping samples on the floor.

This is the reason the internal webserver is now disabled by default - it's just fundamentally broken for serving anything that doesn't fit in the socket's buffer, you'd need to rewrite the whole network layer to be a proper event loop to get it working.

daviessm, 2016-08-04:
Ah yes, true. I forgot I had to hack around just to get the web server enabled on the latest codebase.

closed 2016-08-04 by daviessm