mutability/dump1090 #138: Add option for all aircraft trails (vs selected aircraft)?

contactpjb, 2016-08-06:
Hi,

Wondering if it would be possible to add an option to show aircraft trails for all visible aircraft (not just the selected one), or alternatively all trails since the browser window was opened?

Thanks,
Peter

imesg, 2016-08-27:
Would like this as well.

Thanks,
Gene

samwathegreat, 2016-12-11:
Agreed.  I would prefer an option to show all trails and either have a default or user-defined delay (seconds) for clearing the trail data since the aircraft was last seen.  I'm loving the new features (just updated a 2015 release).  Thanks!

Romeo-Golf, 2016-12-21:
yes this would be nice, firstly to have the option to show the trail for every aircraft's entire flight.. or at least as much as was seen by dump1090...

And if this could be further enhanced to give the option to display trails for all aircraft, but limited to a user defined threshold of either time or distance :)

inpatientia, 2018-07-03:
Here's a quick hack that may achieve the desired effect.  It enables trails for all aircraft currently being tracked without having them clutter the screen forever.

in file /usr/share/dump1090-mutabilityscript.js, line 908
	comment out one line so that selecting and de-selecting planes won't clear the traces:
`//Planes[SelectedPlane].clearLines();`

in file /usr/share/dump1090-mutability/planeObject.js, at line 451
	comment out two lines so that all aircraft show trails
	`//if (!this.selected)`
	`//	return;`

It seems to work fine, but no testing has be done to validate long term stability.  And, of course, extra credit goes to whomever incorporates this as an option enabled/disabled through the UI.

closed 2019-09-14 by mutability