mutability/dump1090 #140: dump1090 don't create any JSON files

gege2b, 2016-08-20:
Hi there

I freshly installed the latest commit of dump1090 (building from source, following the doc)
All went fine, dump1090 runs fine, I can display the map on my browser but I have a popup saying "Problem fetching data from dump1090.
AJAX call failed (error: Not Found). Maybe dump1090 is no longer running?"

So, Dump1090 is running, but he's not creating any json files.
I tried to run it as root : same issue
I tried to give 777 permission on a custom directory (ugly, I know) : same problem

He just refuse to create the json files.
I had this issue on two different computer : a PC running Debian Stretch and a rPI running a freshly installed Jessie

Logs didn't show anything so I can't help you much right now. Please ask if you need more infos

regards

gege

mutability, 2016-08-20:
What are the command line args you are using?

gege2b, 2016-08-20:
--metric --interactive --lat xx.xxxx --lon xx.xxxx 
I also tried with --net 
and I tried overriding default config using --write-json pointing at the default /run/dump1090-mutability dir (whitch I had to manually create because he wasn't existing)

config file was the default one

mutability, 2016-08-21:
If started by hand, there is no default for --write-json. You need to specify it and point it somewhere writable.

mutability, 2016-08-21:
With current master:

```
oliver@travel:~/git/dump1090$ mkdir /tmp/test; ./dump1090 --net-only --write-json /tmp/test
Sun Aug 21 01:42:51 2016 BST  dump1090-mutability v1.15-dev-236-gbc8ba5b starting up.
Net-only mode, no RTL device or file open.
^CSun Aug 21 01:42:53 2016 BST  Caught SIGINT, shutting down..

Sun Aug 21 01:42:53 2016 BST  Normal exit.
oliver@travel:~/git/dump1090$ ls /tmp/test
aircraft.json  history_0.json  receiver.json  stats.json
```

So it looks fine to me. I will need a more specific report to do anything more with this.

closed 2016-08-21 by mutability

gege2b, 2016-08-21:
Well, it's OK...
I didn't know that there is no default for write-json when started by hand...
And as the only test I did specifying it on the command line didn't work, I didn't dig further

thank you and sorry for this non-issue