mutability/dump1090 #145: Makefile breaks raspbian build

kevinbr100, 2016-09-19:
this makefile change fa91b24, breaks building on raspbian, builds ok with original line

-CFLAGS+=-std=c11 -D_BSD_SOURCE=1 -D_POSIX_C_SOURCE=200809L
 +CFLAGS+=-std=c11 -D_DEFAULT_SOURCE

4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

gcc version 4.7.2 (Debian 4.7.2-5+rpi1)

mutability, 2016-09-19:
Upgrade to jessie, I don't really plan to continue supporting wheezy

mutability, 2016-09-19:
Or if you have a fix that builds cleanly in C11 mode on both jessie and wheezy, and does not trigger warnings on newer libc (see issue #142), let me know.

closed 2016-09-19 by mutability

kevinbr100, 2016-09-20:
No worries, I know it's difficult to support a large number of
platforms/OSes, thought it may have been unintentional consequence