mutability/dump1090 #146: No web frontpage with actual dev version

JohnDoeIsKIng, 2016-09-20:
Hi there,

with a fresh checkout of dump1090 all runs well if I start dump1090 without any parameters, i.e. the decoder shows flights on the console. But if I want to open the web frontpage (http://ipoftheraspi:8080), I'm coming out with a 'disconnection error'. Is it possible that this comes from the change to open street map instead of google maps ? Which error logs should I provide (/var/log/dump1090-mutability.log is empty) ?
Regards,

JD.

mutability, 2016-09-20:
The internal webserver has been removed (it doesn't work correctly for the larger file sizes that are now required, and repairing it is a lot of work); you need to use an external webserver to serve the map.

mutability, 2016-09-20:
note that the debian package sets this up to use lighttpd to serve the map automatically, if you're building/running it another way you will need to arrange for something similar to what the package does.

closed 2016-09-20 by mutability

JohnDoeIsKIng, 2016-09-20:
Okay, thanks for pointing that out.
Alternatively I'm using the commit I wrote in the post.
Could you eventually provide a small manual how to use a external webserver ?
How do I sent the data from the raspi external server ?

mutability, 2016-09-20:
Sure, you can use an old commit but it's known to have bugs.

See the debian packaging for an example of using an external webserver.

JohnDoeIsKIng, 2016-09-20:
But I'm actually running a lighttpd on the raspi ... ?

mutability, 2016-09-20:
Then it should be working fine and you should have a map at http://pi/dump1090/

If that doesn't work, what fails?

JohnDoeIsKIng, 2016-09-20:
With the newest version I have the map at http://ipofthepi/dump1090, but no planes shown and the ajax-error on the lower right saying connection broken ...

mutability, 2016-09-20:
So the external webserver config is fine but dump1090 is not writing json files to /run/dump1090. How have you configured dump1090?

mutability, 2016-09-20:
Actually I guess from the other symptoms (no log files, talking about command line options) you are starting dump1090 by hand, so you probably have not passed a --write-json option.

JohnDoeIsKIng, 2016-09-20:
You're wright, I'm starting by hand with a 'screen' . I've tried with --write-json /usr/share/lighttpd, but with no luck.

mutability, 2016-09-20:
Look at the lighttpd config that gets installed. It expects to find the json files in /run/dump1090. Either put them there or change the config.

mutability, 2016-09-20:
nb. I suggest you put the json files on tmpfs as they are rewritten constantly.

JohnDoeIsKIng, 2016-09-20:
That 's the lighttpd.conf as it is installed:
[CODE]
server.modules = (
        "mod_access",
        "mod_alias",
        "mod_compress",
        "mod_redirect",
#       "mod_rewrite",
)

server.document-root        = "/var/www/html"
server.upload-dirs          = ( "/var/cache/lighttpd/uploads" )
server.errorlog             = "/var/log/lighttpd/error.log"
server.pid-file             = "/var/run/lighttpd.pid"
server.username             = "www-data"
server.groupname            = "www-data"
server.port                 = 80

index-file.names            = ( "index.php", "index.html", "index.lighttpd.html" )
url.access-deny             = ( "~", ".inc" )
static-file.exclude-extensions = ( ".php", ".pl", ".fcgi" )

compress.cache-dir          = "/var/cache/lighttpd/compress/"
compress.filetype           = ( "application/javascript", "text/css", "text/html", "text/plain" )

# default listening port for IPv6 falls back to the IPv4 port
include_shell "/usr/share/lighttpd/use-ipv6.pl " + server.port
include_shell "/usr/share/lighttpd/create-mime.assign.pl"
include_shell "/usr/share/lighttpd/include-conf-enabled.pl"
[/CODE]

What path for the json output should I use ? Tried /var/www/html with no luck ...

mutability, 2016-09-20:
/run/dump1090, like I said in my previous comment.

I keep saying this, I don't know how else to say this: Look at the config that the package installs. It's in /etc/lighttpd/conf-available/

JohnDoeIsKIng, 2016-09-20:
But how do I use the  /etc/lighttpd/conf-available/89-dump1090.conf with the lighttpd on the raspi ?
Please excuse the questions, I'm not that familiar with raspian/ubuntu ...

mutability, 2016-09-20:
I suggest you just use the package scripts to start dump1090 if you are out of your depth configuring it by hand, it will start dump1090 with the right args.

JohnDoeIsKIng, 2016-09-20:
Okay, I've set up a new Raspian and installed the stable debian package (1.14) as you suggested.
Than I built a deb-package from the actual sources (1.15) and installed that.
Now everything works fine.
Thanx for you patience.