mutability/dump1090 #147: Dev Version (1.15) collides with piaware 3.0.5 - lighttpd won't start

JohnDoeIsKIng, 2016-09-22:
Greetings,

I've tried to install the actual piaware version as explained here ->[https://de.flightaware.com/adsb/piaware/install](url).
As a result, my running dump1090 won't be getting any adsb-data any more (Ajax-Error).
This is the installation log of piaware:
`sudo apt-get install piaware
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  dump1090-fa itcl3 tcl tcl-tls tcl8.6 tcllib tclx8.4
Suggested packages:
  itcl3-doc tcl-tclreadline tclx8.4-doc
The following NEW packages will be installed:
  dump1090-fa itcl3 piaware tcl tcl-tls tcl8.6 tcllib tclx8.4
0 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
Need to get 6,717 kB of archives.
After this operation, 27.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://flightaware.com/adsb/piaware/files/packages/ jessie/piaware dump1090-fa armhf 3.0.5 [597 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian/ jessie/main itcl3 armhf 3.4.1-2 [42.5 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian/ jessie/main tcl8.6 armhf 8.6.2+dfsg-2 [122 kB]
Get:4 http://mirrordirector.raspbian.org/raspbian/ jessie/main tcl armhf 8.6.0+8+b1 [5,508 B]
Get:5 http://mirrordirector.raspbian.org/raspbian/ jessie/main tcl-tls armhf 1.6+dfsg-3 [57.7 kB]
Get:6 http://mirrordirector.raspbian.org/raspbian/ jessie/main tcllib all 1.16-dfsg-2 [3,950 kB]
Get:7 http://flightaware.com/adsb/piaware/files/packages/ jessie/piaware piaware armhf 3.0.5 [1,864 kB]
Get:8 http://mirrordirector.raspbian.org/raspbian/ jessie/main tclx8.4 armhf 8.4.1-1 [78.5 kB]
Fetched 6,717 kB in 6s (1,071 kB/s)                                            
Selecting previously unselected package itcl3:armhf.
(Reading database ... 121184 files and directories currently installed.)
Preparing to unpack .../itcl3_3.4.1-2_armhf.deb ...
Unpacking itcl3:armhf (3.4.1-2) ...
Selecting previously unselected package tcl8.6.
Preparing to unpack .../tcl8.6_8.6.2+dfsg-2_armhf.deb ...
Unpacking tcl8.6 (8.6.2+dfsg-2) ...
Selecting previously unselected package tcl.
Preparing to unpack .../tcl_8.6.0+8+b1_armhf.deb ...
Unpacking tcl (8.6.0+8+b1) ...
Selecting previously unselected package tcl-tls.
Preparing to unpack .../tcl-tls_1.6+dfsg-3_armhf.deb ...
Unpacking tcl-tls (1.6+dfsg-3) ...
Selecting previously unselected package tcllib.
Preparing to unpack .../tcllib_1.16-dfsg-2_all.deb ...
Unpacking tcllib (1.16-dfsg-2) ...
Selecting previously unselected package tclx8.4.
Preparing to unpack .../tclx8.4_8.4.1-1_armhf.deb ...
Unpacking tclx8.4 (8.4.1-1) ...
Selecting previously unselected package dump1090-fa.
Preparing to unpack .../dump1090-fa_3.0.5_armhf.deb ...
Unpacking dump1090-fa (3.0.5) ...
Selecting previously unselected package piaware.
Preparing to unpack .../piaware_3.0.5_armhf.deb ...
Unpacking piaware (3.0.5) ...
Processing triggers for man-db (2.7.0.2-5) ...
Processing triggers for systemd (215-17+deb8u5) ...
Setting up itcl3:armhf (3.4.1-2) ...
Setting up tcl8.6 (8.6.2+dfsg-2) ...
Setting up tcl (8.6.0+8+b1) ...
Setting up tcl-tls (1.6+dfsg-3) ...
Setting up tcllib (1.16-dfsg-2) ...
Setting up tclx8.4 (8.4.1-1) ...
Setting up dump1090-fa (3.0.5) ...
Enabling lighttpd integration..
Enabling dump1090-fa: ok
Run /etc/init.d/lighttpd force-reload to enable changes
Restarting lighttpd..
Job for lighttpd.service failed. See 'systemctl status lighttpd.service' and 'journalctl -xn' for details.
invoke-rc.d: initscript lighttpd, action "restart" failed.
Setting up piaware (3.0.5) ...
Reloading rsyslog configuration..
Processing triggers for libc-bin (2.19-18+deb8u6) ...
Processing triggers for systemd (215-17+deb8u5) ...
`
It seems that went something wrong with the configuration of the lighttpd by piaware ...
Any hints are deeply appreciated!
Greets,

JD.

mutability, 2016-09-22:
dump1090-fa and dump1090-mutability don't coexist without some configuration. Both want to configure the same lighttpd settings and lighttpd gets confused by having the same setting set twice, even though the values are the same.

You will need to disable one or the other.

mutability, 2016-09-22:
Note in particular that the piaware instructions say:

> If you don't already have ADS-B receiver software such as dump1090 installed, (... instructions on installing dump1090-fa)

If you do already have ADS-B receiver software installed, then you should have skipped that step.

closed 2016-09-22 by mutability

mutability, 2016-09-22:
I can probably do something about this with update-alternatives, I guess, it's just a pain. The assumption is that if you're using dump1090-mutability then you know what you're doing.

JohnDoeIsKIng, 2016-09-22:
Okay, thanks for the reply. Of course I'm using dump1090-mutability, but I thought that thre is a chance that, regarding only to the decoder part, the two forks could coexist.
I I can do something regarding a patch for the lighttpd-configs, let me know.
Btw: I'm running an older dump1090-mutability-dev.version (1.15), maybe from March, that works okay with piaware 2.1-5 ...

mutability, 2016-09-22:
piaware and dump1090 are two separate things. Of course dump1090-mutability works OK with PiAware 2.1-5. It also works OK with PiAware 3.0.5. It's only if you install dump1090-fa that you will have issues.

JohnDoeIsKIng, 2016-09-22:
I'm confused about the difference between dump1090-fa and piaware ...
If there 's a "single" client independent from dump1090-mutability that enables me to send the data from dump1090-mutability to my flightaware account, please let me know !

mutability, 2016-09-22:
That is exactly what piaware is.

mutability, 2016-09-22:
TBH I suggest you try just using the piaware sdcard image as that has everything preinstalled and preconfigured.

JohnDoeIsKIng, 2016-09-22:
But how comes that on my other raspi dump1090-mutability and piaware 2.1-5 work percetly together and I can feed the data to my fa-account ?

JohnDoeIsKIng, 2016-09-22:
Okay, but I want the nice GUI features of dump1090-mutability (different shapes and altitude dependent colors of planes, FR24-Link-Integration and so on) ! :-)

mutability, 2016-09-22:
Because.. you installed different software there?
Have you tried dump1090-mutability with piaware 3.0.5? No dump1090-fa. You don't need dump1090-fa if you already have a dump1090 installed. Read the piaware install instructions.

mutability, 2016-09-22:
Also, for future support, please try the FA forums or mail adsbsupport@flightaware.com, github is not a good place to get general help with installing packages, it's aimed at actual bug reports.

JohnDoeIsKIng, 2016-09-22:
Okay, thanks. I will post my questions there.
Thanks for your explanations and your patience.

JohnDoeIsKIng, 2016-09-22:
I've found a solution: I've build a piaware package like explained here [https://www.swiftbyte.com/articles/software-defined-radio/setting-up-piaware-and-dump1090-mutability-on-raspbian-ubuntu-or-debian](url)
During the built / installation I had to install a few packages, but after that it runs smoothly.
Best regards,

JD.

mutability, 2016-09-22:
OK, so that would be "Don't install dump1090-fa, just piaware", like I said above.

mutability, 2016-09-22:
I separated out the statcache config in dump1090-fa and I'll do the same in dump1090-mutability at some point, which should mitigate the immediate problem, but having both dump1090-fa and dump1090-mutability installed at the same time still doesn't make much sense because you're going to have to disable one or the other, they can't both use the dongle at the same time.

JohnDoeIsKIng, 2016-09-22:
'OK, so that would be "Don't install dump1090-fa, just piaware", like I said above.'

That 's what I thought to do as I installed the package named piaware ...
Maybe It comes as a collateral with installing the repository ..
So in sum: As I installed the package according to thelink in the first post, it comes to the error I described. To be concrete: I never installed a package called dump1090-fa or something similar, only tried to install the current piaware-version!
Anywhere, it runs. Thanks for replying.

mutability, 2016-09-22:
And yet the package got installed, so you did something that caused it to be installed. I can't really diagnose that without a careful description of what you did and what the results were.

mutability, 2016-09-22:
The piaware package does Recommend: dump1090-fa, so if you installed without --no-recommends it would have been installed too. I guess maybe that should be dropped to Suggests: