mutability/dump1090 #148: Web page issue: Not updating location from gps

meijin007, 2016-09-22:
system: RPI 2 with ArchLinuxArm, latest dump1090 from git.
there is GPS connected to the rpi.
In the config.json there is this documentation: 

> // Center marker. If dump1090 provides a receiver location,
> // that location is used and these settings are ignored.
> 
> SiteShow    = false;           // true to show a center marker
> SiteLat     = 45.0;            // position of the marker
> SiteLon     = 9.0;
> SiteName = "My Radar Site"; // tooltip of the marker

even thou the gps is working (I can see my location on FlightAware site), and SiteShow set to true, the coordinates aren't set to the gps location.

Also - SiteName doesn't show as tip on the site spot.

mutability, 2016-09-22:
dump1090-mutability doesn't have any integration with the GPS location information that piaware collects; use dump1090-fa for that.

mutability, 2016-09-22:
Or you can arrange for dump1090 to be started including the args from /var/cache/piaware/location.env, the details of which you will need to work out yourself as I have no idea how the arch packaging is set up.

closed 2016-09-22 by mutability

anish, 2016-09-22:
@mutability sorry to jump in on the thread, but where does dump1090-fa reside nowadays ? My understanding was that the repo was removed from flightaware

mutability, 2016-09-22:
http://github.com/flightaware/dump1090

anish, 2016-09-22:
@meijin007 try using https://aur.archlinux.org/packages/dump1090-fa-git/ (despite the name, it pulls from this repo)

meijin007, 2016-09-22:
@anish actually, this is the package I am using.

> pacman -Qs dump1090
> local/dump1090-fa-git 1:1.15.r288.g729d91c-1

I found the location where @mutability said, is there a way to send it to dump1090 as argument --lat and --lon ?