mutability/dump1090 #152: rings not centered.

arnauldb, 2016-10-07:
Hello,

thank you for dump1090.

I have a problem, how can I center the ring overlay or other overlays, heatmap, on my location ? I am in Paris but the overlays are centered on the Netherlands...

thank you,
[image]

mutability, 2016-10-07:
You've made modifications to the dump1090 map UI. I can't provide support for those.

closed 2016-10-07 by mutability

Romeo-Golf, 2016-12-23:
Not sure if it's any help but I notice the distances are in the ~400km range which is about the distance from Paris to Amsterdam... 

Could it be that dump1090 has somehow been started with the -lat and -lon arguments containing coordinates from a location in the Netherlands? :)

arnauldb, 2016-12-23:
thanks Romeo-Golf, you are right I started dump1090 with the wrong coordinates.