mutability/dump1090 #159: stats.json signal info only available on over sample

coolacid, 2016-10-25:
It appears that the code to generate signal power stats is only available in the 2.4 oversample file. 

Not sure if this is an oversight or technically not possible. Took me a while (both searching web and issues/code) to find the issue. 

To make it easier for people to find this ticket, some extra keywords:
peak_signal
strong_signals

mutability, 2016-10-25:
I don't understand the problem; the 2.4MHz is the only demodulator now.

coolacid, 2016-10-25:
I was trying to figure out why I'm not getting as much signal as I thought I should be - part of this troubleshooting was changing the variable OVERSAMPLE in /etc/default/dump1090-mutability to "no" this caused the signals data to not be written to the json file.

mutability, 2016-10-25:
OVERSAMPLE does exactly nothing in the current code and scripts, so you must be running an older version.

closed 2016-10-25 by mutability