mutability/dump1090 #162: Failing to build on RPi2

ant-thomas, 2016-11-05:
After not updating my RPi2 system for a long time I've decided to build dump1090 again to update everything. Trying to build from scratch and I'm getting the following...

`root@pi:~/dump1090-mut-2016/dump1090# dpkg-buildpackage -b
dpkg-buildpackage: source package dump1090-mutability
dpkg-buildpackage: source version 1.15~dev
dpkg-buildpackage: source changed by Oliver Jowett <oliver@mutability.co.uk>
dpkg-buildpackage: host architecture armhf
 dpkg-source --before-build dump1090
 debian/rules clean
dh clean 
   dh_testdir
   dh_auto_clean
	make -j1 clean
make[1]: Entering directory '/root/dump1090-mut-2016/dump1090'
rm -f *.o compat/clock_gettime/*.o compat/clock_nanosleep/*.o dump1090 view1090 faup1090 cprtests crctests
make[1]: Leaving directory '/root/dump1090-mut-2016/dump1090'
   dh_clean
	rm -f debian/dump1090-mutability.substvars
	rm -f debian/dump1090-mutability.*.debhelper
	rm -rf debian/dump1090-mutability/
	rm -f debian/*.debhelper.log
	rm -f debian/files
	find .  \( \( -type f -a \
	        \( -name '#*#' -o -name '.*~' -o -name '*~' -o -name DEADJOE \
		 -o -name '*.orig' -o -name '*.rej' -o -name '*.bak' \
		 -o -name '.*.orig' -o -name .*.rej -o -name '.SUMS' \
		 -o -name TAGS -o \( -path '*/.deps/*' -a -name '*.P' \) \
		\) -exec rm -f {} \; \) -o \
		\( -type d -a -name autom4te.cache -prune -exec rm -rf {} \; \) \)
	rm -f *-stamp
 debian/rules build
dh build 
   dh_testdir
   dh_auto_configure
   debian/rules override_dh_auto_build
make[1]: Entering directory '/root/dump1090-mut-2016/dump1090'
dh_auto_build -- 'EXTRACFLAGS=-DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\"'
	make -j1 "EXTRACFLAGS=-DHTMLPATH=\\\"/usr/share/dump1090-mutability/html\\\" -DMODES_DUMP1090_VARIANT=\\\"dump1090-mutability\\\""
make[2]: Entering directory '/root/dump1090-mut-2016/dump1090'
gcc -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fPIE -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -std=c11 -D_DEFAULT_SOURCE `pkg-config --cflags librtlsdr` -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c dump1090.c -o dump1090.o
cc1: error: unrecognized command line option ‘-std=c11’
Makefile:49: recipe for target 'dump1090.o' failed
make[2]: *** [dump1090.o] Error 1
make[2]: Leaving directory '/root/dump1090-mut-2016/dump1090'
dh_auto_build: make -j1 EXTRACFLAGS=-DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" returned exit code 2
debian/rules:21: recipe for target 'override_dh_auto_build' failed
make[1]: *** [override_dh_auto_build] Error 2
make[1]: Leaving directory '/root/dump1090-mut-2016/dump1090'
debian/rules:30: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
root@pi:~/dump1090-mut-2016/dump1090#`

Just tried to build on an x86_64 system and it worked fine first time (not cross-compile).

I have done `apt-get update && apt-get upgrade && apt-get dist-upgrade` - running Raspbian Wheezy.

Am I missing a new package or dependency?
GCC version - `gcc (Debian 4.6.3-14+rpi1) 4.6.3`

ant-thomas, 2016-11-05:
And just regular make
```
root@pi:~/dump1090-mut-2016/dump1090# make
gcc -DMODES_DUMP1090_VERSION=\"v1.15-dev-305-gb957166\" -O2 -g -Wall -Werror -W -std=c11 -D_DEFAULT_SOURCE `pkg-config --cflags librtlsdr`  -c dump1090.c -o dump1090.o
cc1: error: unrecognized command line option ‘-std=c11’
Makefile:49: recipe for target 'dump1090.o' failed
make: *** [dump1090.o] Error 1
root@pi:~/dump1090-mut-2016/dump1090# 
```

OrbitTheSun, 2016-11-09:
You need to update your gcc to version 4.9. See
http://stackoverflow.com/questions/25147363/how-to-install-g-4-9-on-debian-wheezy-armel
for hints to do that.

mutability, 2017-02-08:
Alternatively you might want to look at what dump1090-fa does to built on wheezy

https://github.com/flightaware/dump1090/blob/master/debian-wheezy/rules

tl;dr: pass -std=gnu1x

mutability, 2017-02-09:
In general I don't test dump1090-mutability on wheezy; maybe it works, maybe it doesn't.

dump1090-fa takes more care as it needs to support some legacy wheezy images.

If you have patches for wheezy for dump1090-mutability I'm happy to take a PR.

closed 2017-02-09 by mutability