mutability/dump1090 #164: How to send the selected addr back to the dump1090?

dragonyzl, 2016-11-08:
Hi, Dear Oliver Jowett and others:
    Thanks for your great improvement of dump1090.
    Now I want to use dump1090 to guide my telescope to point to the airplane I selected in the browser, the airplane's address need to be sent back to the dump1090 embedded webserver to filter out the lat and lon of selected airplane form huge decoded messages interactively.  At the beginning, I think it is easy to write a simple text file using JavaScript based on the code in public_html, it is not allowed because of security. I googled and found maybe the libwebsocket is useful. but I am not familiar with the web programming and confused about the examples in the libwebsocket. Could someone please give me some hint? Thanks in advance!

mutability, 2016-11-08:
You might want to just read aircraft.json directly; this contains all the current aircraft positions.

mutability, 2016-11-08:
Oh, I guess you want to hook it up to the action of selecting the aircraft in the browser so you can just select an aircraft and have that drive the telescope? That is more complex and I don't have a simple solution for you.

nb. I don't support the embedded webserver any more.

dragonyzl, 2016-11-08:
@mutability, Ooh, May be I can " echo selectedAddr > file"  manually, it looks so foolish! 
Thanks!

dragonyzl, 2016-12-05:
I have solved the problem using FileSaver.js. it looks nice.

closed 2016-12-05 by dragonyzl