mutability/dump1090 #165: Problem fetching data from dump1090

Strykar, 2016-11-23:
Where do I start troubleshooting this error? This is on x64 OpenWRT/LEDE, and dump1090 works fine and reports crafts when run from console. The dump1090 web interface running on port 8080 shows:
```
Problem fetching data from dump1090.
AJAX call failed (parsererror: SyntaxError: Unexpected end of JSON input). Maybe dump1090 is no longer running?
The displayed map data will be out of date.

# ps -e | grep dump1090
14476 ttyS0    00:00:01 dump1090
```
https://github.com/openwrt/packages/blob/cd13ac4ae158cff7293e5385abe5084aeebb6c42/utils/dump1090/Makefile

mutability, 2016-11-23:
Check the javascript console on your browser and/or look at the actual json the browser got.

Strykar, 2016-11-23:
Thanks, I sniffed the conversation with fiddler! It appears the target directory the softlink pointed to was not created, and thus aircraft.json did not exist. Creating it and aircraft.json inside it fixed the issue and crafts now appear on http://router:8080
```
root@apu:/# mkdir /var/run/dump1090
root@apu:/# touch /var/run/dump1090/aircraft.json
root@apu:/# ls -l /usr/share/dump1090/data
lrwxrwxrwx    1 root     root            17 Oct 27 18:59 /usr/share/dump1090/data -> /var/run/dump1090
```
I''ll create an issue with OpenWRT packages.

closed 2016-11-23 by Strykar