mutability/dump1090 #167: Can not display in the browser, why?

dragonyzl, 2016-12-04:
After I upgrade the Ubuntu from 14.04 to 16.04, and recreat the dump1090, the following errors found in browser and no planes displayed in the browser:
ReferenceError: ol is not defined[详细了解]  ol3-layerswitcher.js:9:1
ReferenceError: ol is not defined[详细了解]  script.js:6:5
TypeError: PlanesOrdered is undefined[详细了解]  script.js:873:25

Could someone tell me why?
Thanks in advance!

giomasce, 2016-12-04:
I have the same problem. It appears that the request for file ol3/ol-3.17.1.js fails: the `Content-Length` header is correct, but the response contains no data. All the other requests appear to be fine. I don't understand why that one fails specifically.

mutability, 2016-12-04:
What webserver are you using?

giomasce, 2016-12-04:
The internal one. I just added -DENABLE_WEBSERVER to the compilation flags.

mutability, 2016-12-04:
That is unsupported and known to be broken. Use an external webserver.

closed 2016-12-04 by mutability

dragonyzl, 2016-12-05:
In fact, the internal webserver is good for a standalone program that is suitable for some embedded application.

OKay, I am not familiar with the network app development. May be someone can solve the issue.

mutability, 2016-12-05:
Good except it doesn't actually work for larger files as you have
discovered. It needs a rewrite to fix it and that is not interesting for
me, though I will take patches if someone else wants to do the work

dragonyzl, 2016-12-18:
I found the reason is that the "(" is removed in the file ol3/ol-3.17.1.js, after the "(" is added to the file, everything is OK