mutability/dump1090 #168: Skip the start dump1090 sutomatically question.

jprochazka, 2016-12-08:
When installing the dump1090-mutability package the user is asked if dump1090-mutability should be started automatically. Being that %99.9999 of the time this will be answered by a yes by the user is there a way to skip having to answer this question and have it default to yes so installation continues on uninterrupted?

Romeo-Golf, 2016-12-21:
I would also find this useful ;)

ralin3, 2017-01-27:
If you install debconf-utils, you can bypass it by preseeding the answer:
# echo 'dump1090-mutability     dump1090-mutability/auto-start  boolean true' > /root/dump1090.seed
# debconf-set-selections /root/dump1090.seed
Better to fix it of course, but at least it works for now.

mutability, 2017-01-27:
That is in fact the right way to fix it - prepopulate the answer in the debconf db.

Romeo-Golf, 2017-01-30:
cheers, as suggested the following appears to do trick:

> $ debconf-get-selections | grep "dump1090-mutability/auto-start"
> dump1090-mutability     dump1090-mutability/auto-start  boolean false
>
> $ echo 'dump1090-mutability     dump1090-mutability/auto-start  boolean true' | debconf-set-selections -v
> info: Trying to set 'dump1090-mutability/auto-start' [boolean] to 'true'
> info: Loading answer for 'dump1090-mutability/auto-start'
> 
> $ debconf-get-selections | grep "dump1090-mutability/auto-start"
> dump1090-mutability     dump1090-mutability/auto-start  boolean true
 >

:)

closed 2017-02-09 by mutability