mutability/dump1090 #177: Can't compile on CentOS 7

jmceara, 2017-01-27:
Hi. I'm trying to compile in CentOS 7, but I'm getting a lot of errors....mostly of implicity declaration of functions and not defined constants (CLOCK_REALTIME, for example). All needed libraries are included. GCC version 4.8.5.

mutability, 2017-01-27:
You need to tell me what you ran and what the errors were; a vague description doesn't give me anything I can diagnose.

jmceara, 2017-01-27:
`[jmaurin@skynet dump1090]$ make
gcc -DMODES_DUMP1090_VERSION=\"v1.15-dev-308-gccd8fd5\" -O2 -g -Wall -Wmissing-declarations -W -std=c11 -D_DEFAULT_SOURCE `pkg-config --cflags librtlsdr`  -c dump1090.c -o dump1090.o
dump1090.c: In function ‘log_with_timestamp’:
dump1090.c:73:5: warning: implicit declaration of function ‘localtime_r’ [-Wimplicit-function-declaration]
     localtime_r(&now, &local);
     ^
dump1090.c: At top level:
dump1090.c:103:5: warning: no previous declaration for ‘getTermRows’ [-Wmissing-declarations]
 int getTermRows() {
     ^
dump1090.c:110:6: warning: no previous declaration for ‘sigWinchCallback’ [-Wmissing-declarations]
 void sigWinchCallback() {
      ^
dump1090.c: In function ‘start_cpu_timing’:
dump1090.c:122:5: warning: implicit declaration of function ‘clock_gettime’ [-Wimplicit-function-declaration]
     clock_gettime(CLOCK_THREAD_CPUTIME_ID, start_time);
     ^
dump1090.c:122:19: error: ‘CLOCK_THREAD_CPUTIME_ID’ undeclared (first use in this function)
     clock_gettime(CLOCK_THREAD_CPUTIME_ID, start_time);
                   ^
dump1090.c:122:19: note: each undeclared identifier is reported only once for each function it appears in
dump1090.c: In function ‘end_cpu_timing’:
dump1090.c:128:19: error: ‘CLOCK_THREAD_CPUTIME_ID’ undeclared (first use in this function)
     clock_gettime(CLOCK_THREAD_CPUTIME_ID, &end_time);
                   ^
dump1090.c: At top level:
dump1090.c:138:6: warning: no previous declaration for ‘modesInitConfig’ [-Wmissing-declarations]
 void modesInitConfig(void) {
      ^
dump1090.c: In function ‘modesInitConfig’:
dump1090.c:148:5: warning: implicit declaration of function ‘strdup’ [-Wimplicit-function-declaration]
     Modes.net_input_raw_ports     = strdup("30001");
     ^
dump1090.c:148:35: warning: assignment makes pointer from integer without a cast [enabled by default]
     Modes.net_input_raw_ports     = strdup("30001");
                                   ^
dump1090.c:149:35: warning: assignment makes pointer from integer without a cast [enabled by default]
     Modes.net_output_raw_ports    = strdup("30002");
                                   ^
dump1090.c:150:35: warning: assignment makes pointer from integer without a cast [enabled by default]
     Modes.net_output_sbs_ports    = strdup("30003");
                                   ^
dump1090.c:151:35: warning: assignment makes pointer from integer without a cast [enabled by default]
     Modes.net_input_beast_ports   = strdup("30004,30104");
                                   ^
dump1090.c:152:35: warning: assignment makes pointer from integer without a cast [enabled by default]
     Modes.net_output_beast_ports  = strdup("30005");
                                   ^
dump1090.c: At top level:
dump1090.c:166:6: warning: no previous declaration for ‘modesInit’ [-Wmissing-declarations]
 void modesInit(void) {
      ^
dump1090.c: In function ‘modesInit’:
dump1090.c:233:13: warning: implicit declaration of function ‘le16toh’ [-Wimplicit-function-declaration]
             Modes.maglut[le16toh((i*256)+q)] = (uint16_t) round(sqrtf(magsq) * 65535.0);
             ^
dump1090.c: At top level:
dump1090.c:277:5: warning: no previous declaration for ‘modesInitRTLSDR’ [-Wmissing-declarations]
 int modesInitRTLSDR(void) {
     ^
dump1090.c:390:6: warning: no previous declaration for ‘rtlsdrCallback’ [-Wmissing-declarations]
 void rtlsdrCallback(unsigned char *buf, uint32_t len, void *ctx) {
      ^
dump1090.c: In function ‘rtlsdrCallback’:
dump1090.c:455:19: error: ‘CLOCK_REALTIME’ undeclared (first use in this function)
     clock_gettime(CLOCK_REALTIME, &outbuf->sysTimestamp);
                   ^
dump1090.c: At top level:
dump1090.c:490:6: warning: no previous declaration for ‘readDataFromFile’ [-Wmissing-declarations]
 void readDataFromFile(void) {
[... 52 lines cut ...]
                                          ^
dump1090.c:995:41: warning: assignment makes pointer from integer without a cast [enabled by default]
             Modes.net_input_beast_ports = strdup(argv[++j]);
                                         ^
dump1090.c:998:36: warning: assignment makes pointer from integer without a cast [enabled by default]
             Modes.net_bind_address = strdup(argv[++j]);
                                    ^
dump1090.c:1010:40: warning: assignment makes pointer from integer without a cast [enabled by default]
             Modes.net_output_sbs_ports = strdup(argv[++j]);
                                        ^
dump1090.c:1088:28: warning: assignment makes pointer from integer without a cast [enabled by default]
             Modes.html_dir = strdup(argv[++j]);
                            ^
dump1090.c:1091:28: warning: assignment makes pointer from integer without a cast [enabled by default]
             Modes.json_dir = strdup(argv[++j]);
                            ^
dump1090.c:1169:13: warning: implicit declaration of function ‘usleep’ [-Wimplicit-function-declaration]
             usleep(100000);
             ^
dump1090.c:1188:31: error: ‘CLOCK_REALTIME’ undeclared (first use in this function)
                 clock_gettime(CLOCK_REALTIME, &ts);
                               ^
make: *** [dump1090.o] Error 1
[jmaurin@skynet dump1090]$
`

mutability, 2017-01-27:
I guess you have a libc that doesn't understand _DEFAULT_SOURCE.

Also your gcc appears to turn on some extra warnings by defaults which is going to cause you problems.

I don't try compiling on CentOS/RH here; you'll need to work this one out yourself I'm afraid. If you have a Makefile patch I'm happy to take that.

closed 2017-02-09 by mutability