mutability/dump1090 #192: Issue: ./dump1090 --stats --ifile testfiles/modes1.bin

gunawanw9, 2017-02-15:
Is this normal or somewhat ...?

Wed Feb 15 17:56:02 2017 WIB  dump1090-mutability v1.15~dev starting up.
Using sample converter: UC8, integer/table path
Wed Feb 15 17:56:02 2017 WIB  Waiting for receive thread termination

Statistics: Wed Feb 15 17:56:02 2017 WIB - Wed Feb 15 17:56:02 2017 WIB
Local receiver:
  356868 samples processed
  0 samples dropped
  0 Mode A/C messages received
  2181 Mode-S message preambles received
    1218 with bad message format or invalid CRC
    963 with unrecognized ICAO address
    0 accepted with correct CRC
  -14.2 dBFS noise power
  0 messages with signal power above -3dBFS
0 total usable messages
0 surface position messages received
0 airborne position messages received
0 global CPR attempts with valid positions
0 global CPR attempts with bad data
  0 global CPR attempts that failed the range check
  0 global CPR attempts that failed the speed check
0 global CPR attempts with insufficient data
0 local CPR attempts with valid positions
  0 aircraft-relative positions
  0 receiver-relative positions
0 local CPR attempts that did not produce useful positions
  0 local CPR attempts that failed the range check
  0 local CPR attempts that failed the speed check
0 CPR messages that look like transponder failures filtered
0 non-ES altitude messages from ES-equipped aircraft ignored
0 unique aircraft tracks
0 aircraft tracks where only one message was seen
CPU load: 106.7%
  10 ms for demodulation
  6 ms for reading from USB
  0 ms for network input and background tasks
Wed Feb 15 17:56:02 2017 WIB  Normal exit.

any clue how to fix this issue? Thank you.

mutability, 2017-02-15:
What behaviour did you expect? How did you capture the input file?

gunawanw9, 2017-02-15:
that's input file already there, have you try that test file again recently with dump1090-mutability v1.15~dev?

mutability, 2017-02-15:
Oh, that's an old 2MHz testfile, it won't work with the 2.4MHz demodulator. I'll remove it.

commit 00349624 (2017-02-15): Remove an ancient 2MHz capture.

Fixes #192
closed 2017-02-15 by mutability