mutability/dump1090 #193: Use the internal web server

bogdanr, 2017-04-08:
How do I actually use the internal web server. I only run thins for short periods of time on localhost and I would like to use the internal webserver for that because it would be easier. How do I actually do that?

strix-technica, 2017-04-23:
Add the line

    CPPFLAGS+=-DENABLE_WEBSERVER

to the top of the Makefile.

But, as of this writing, there's something wrong with the javascript.  I have not yet worked out what, but Chrome reports problems with ol3 and jQuery.  Perhaps this is related to google's maps API policy changes.

mutability, 2017-04-23:
There is nothing wrong with the JavaScript. You will just be hitting the
same set of bugs that lead me to disable the internal webserver. I will
remove that code entirely when I have some time - it is beyond repair
without rewriting the whole network layer, which isn't something I plan on
doing any time soon.

closed 2017-04-23 by mutability

strix-technica, 2017-04-23:
@mutability,

Sounds like there are two separate issues here: 1) use of Google Maps and API keys, and 2) apparently buggy internal httpd.  For the latter, I guess it makes sense to throw out the internal code when there are perfectly good alternatives (no reason to reinvent [and maintain] the wheel).  

But the new code doesn't use Google Maps anymore, which is a shame if it does actually still work (as it seems to for me, and my installation is new so I can't see how it would be grandfathered).

I'll give lighttpd a try and see whether the problems persist.  If I have trouble, I'll let you know in a new issue.

mutability, 2017-04-23:
The google maps API doesn't work consistently for everyone; you got lucky.

mutability, 2017-04-23:
(also, it's hardly "new code" - Google's policy change and the resulting switch to the OL3 API happened 9 months ago!)

mutability, 2017-04-23:
also, also - the Google API doesn't work at all in places where Google is blocked (notably China); this is something of a showstopper even if the API key issue could be fixed somehow.

strix-technica, 2017-04-23:
Thanks for taking the time to explain :)

I've been fooling around with adsbSCOPE now that i finally got some form of dump1090 working on my pi; even so, the pi-based webserver is still useful because adsbSCOPE doesn't show range information quite like the web interface does.  

I shall now try to get HEAD working with a highttpd and see what the new web interface looks like.