mutability/dump1090 #195: interactive TTL ignored

philipz-59, 2017-04-20:
How to use this parameter? When specifying ./dump1090 --interactive --interactive-ttl 50000,
aircrafts are discarded after 300 sec, whereas dump1090_antirez works as desired..
Regards!

mutability, 2017-04-21:
Are you trying to set a timeout of 50s or 50,000s?

philipz-59, 2017-04-21:
As described in help, trying to set a timeout to 50000 seconds.
Now I bisected problem, everything above 300s is discarded..

mutability, 2017-04-21:
The main tracking logic has a hardcoded timeout of 300s; if you want to display aircraft older than that, you'll need to increase that timeout (TRACK_AIRCRAFT_TTL in track.h) and rebuild.

philipz-59, 2017-04-21:
Ok. And where can I setup prolonged visibility for web frontend? Default is also 60 seconds.
Regards!

philipz-59, 2017-04-21:
aircraft visibility.. ;)

mutability, 2017-04-21:
There's no one place, you'll have to dig through the javascript

philipz-59, 2017-05-05:
Consider it closed, regards!

closed 2019-09-14 by mutability