mutability/dump1090 #196: Problem with --net

itundret, 2017-04-22:
I am running this on Ubuntu. Software is up and running and i can see all the flight in the terminal. But the localhost:8080 access is not working. I have tried to restart it but that don't fix the problem. 
When i checked if the port 8080 is open is not on, my machine.

The original dump 1090 is working fine with local access.  Bug?

mutability, 2017-04-22:
The internal webserver was disabled long ago. It was unreliable and
insecure. Use an external webserver. The Debian packaging provides examples.

CharlieAt, 2017-05-01:
Hi Oliver, in spite of the internal being buggy etc, I wanted it to run at home for pure local observation (dump1090 on a pi zero w). 
After enabling the internal server, the webpage would only partially load and mainly I would have the spinner running and that was that.
It seems that the unix write fn in handleHTTPRequest does as in the man page and will not always write all the data. By adding the attached write_all above handleHTTPRequest and replacing the two writes with write_all's, the internal server is doing what I need and pretty reliably. 
```
static int write_all(int fd, char *buf, int len) {
    int i, wr=0;
    while (wr < len) {
        i = write(fd,&buf[wr], len-wr);
        if (i<0) {
            if (errno!=EAGAIN) return i;
            usleep(1000);
        } else {
            wr += i;
        } 
    }
    return wr;
} 
```
nowhere near production code as the 1ms wait may influence other senders (have not dug that far into the rest of the server architecture yet), but 'fits where it touches' ;-)
By the sounds of it you would like to eradicate the internal server totally, so no push request. Hopefully anyone else struggling with this problem can help themselves.

mutability, 2017-05-01:
If it works for you, more power to you, but I'm not going to support it and it's a fundamentally broken solution because you have stalled all demodulation while you wait for the client to consume the data, which could take _minutes_

philipz-59, 2017-05-05:
Btw, is there some easy solution how to adjust script.js for acessing web frontend directly from filesystem?
E.g. from chrome   file:///myfilesystem/dump1090_mutability/public_html/gmap.html

Sadly, I do not speak Ajax & js.. Regards!

e2jk, 2017-07-14:
Maybe the executable should show a warning/error messages when invoked with the --net parameter, to make [new] users aware that this option is not supported anymore, and that they should use an external webserver (it had me puzzled a bit until I stumbled upon this issue).
By the way, using an external webserver is much nicer, as when you refresh the page the previous positions of the aircraft are shown - with the internal webserver the traces would only start from the moment you refreshed the page, now with the external webserver the data is available even if the web browser didn't yet load the page.

closed 2019-09-14 by mutability