mutability/dump1090 #200: Does not compile

meijin007, 2017-06-14:
I am having an error:
interactive.c:143:44: error: '%03d' directive output may be truncated writing between 3 and 9 bytes into a region of size 5 [-Werror=format-truncation=]
                         snprintf(strFl,6,"F%03d",((a->altitude+50)/100));
                                            ^~~~
interactive.c:143:42: note: directive argument in the range [-21474836, 21474836]
                         snprintf(strFl,6,"F%03d",((a->altitude+50)/100));
                                          ^~~~~~~
In file included from /usr/include/stdio.h:939:0,
                 from dump1090.h:65,
                 from interactive.c:50:
/usr/include/bits/stdio2.h:64:10: note: '__builtin___snprintf_chk' output between 5 and 11 bytes into a destination of size 6
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:49: interactive.o] Error 1

any ideas?

mutability, 2017-06-14:
What compiler?

meijin007, 2017-06-14:
GCC.
I am building with Arch linux AUR pacakge.

mutability, 2017-06-14:
gcc what version?

It's kind of a useless warning - snprintf already handles the truncation itself in the way which we expect. I would be surprised if gcc enabled that warning as part of -Wall. Perhaps the AUR (which I have no information on) passes extra options?

mutability, 2017-06-14:
Also, you should raise this with the maintainer of the package if it is a build problem with their package.

meijin007, 2017-06-14:
As I wrote, gcc 7.1.1
I also raised it in the AUR package page, but it happen with 2 packages (building the same software...).
This is the build command in the aur build:
`make all faup1090 EXTRACFLAGS=-DHTMLPATH=\\\"/usr/share/dump1090/html\\\`

mutability, 2017-06-14:
Yeah, you edited in the version after my comment, good job.

meijin007, 2017-06-14:
Sorry, didn't saw your comment while I was editing

mutability, 2017-06-14:
I test with 5.4.0 at the moment so it's possible that it's just a newer gcc thing which has changed the set of warnings enabled by -Wall. 

As I don't have 7.1.1 or an Arch environment on hand to test against I will need a PR that you (or the package maintainer) have tested to fix this.

meijin007, 2017-06-14:
it seems that there is a need to add 
`CFLAGS += `-Wno-format-truncation`
if gcc version is greater then 7.
I am not familiar enough with Makefile syntax to add a condition, but without condition it seems to solve the problem

meijin007, 2017-06-20:
Added a PR which fix it...

PiRad108, 2018-03-04:
I get the same error  
`make
gcc -DMODES_DUMP1090_VERSION=\"v1.15-dev-330-gcc53c90\" -O2 -g -Wall -Werror -W -std=c11 -D_DEFAULT_SOURCE `pkg-config --cflags librtlsdr libusb-1.0`  -c dump1090.c -o dump1090.o
gcc -DMODES_DUMP1090_VERSION=\"v1.15-dev-330-gcc53c90\" -O2 -g -Wall -Werror -W -std=c11 -D_DEFAULT_SOURCE  -c anet.c -o anet.o
gcc -DMODES_DUMP1090_VERSION=\"v1.15-dev-330-gcc53c90\" -O2 -g -Wall -Werror -W -std=c11 -D_DEFAULT_SOURCE  -c interactive.c -o interactive.o
interactive.c: In function ‘interactiveShowData’:
interactive.c:143:44: error: ‘%03d’ directive output may be truncated writing between 3 and 9 bytes into a region of size 5 [-Werror=format-truncation=]
                         snprintf(strFl,6,"F%03d",((a->altitude+50)/100));
                                            ^~~~
interactive.c:143:42: note: directive argument in the range [-21474836, 21474836]
                         snprintf(strFl,6,"F%03d",((a->altitude+50)/100));
                                          ^~~~~~~
interactive.c:143:25: note: ‘snprintf’ output between 5 and 11 bytes into a destination of size 6
                         snprintf(strFl,6,"F%03d",((a->altitude+50)/100));
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:49: recipe for target 'interactive.o' failed
make: *** [interactive.o] Error 1
`My  installer is 
gcc (debian 7.3.0-5) 7.3.0

I am running Kali3 on my laptop

closed 2019-09-14 by mutability