mutability/dump1090 #202: Simple stats page

kevdel, 2017-06-20:
Any interest in a simple consolidated page with stats so that you could pull the latest info into another app?

E.g http://server/stats

Produces a delimited plain text list such as
Total aircraft|adsb aircraft|mlat aircraft|messages per sec|max distance

Other elements like closest aircraft or furthest aircraft could be extracted from the status page

Just a thought...

mutability, 2017-06-20:
Stats like this are already available as json, can you use that?

kevdel, 2017-06-20:
Yes, i was planning to use the content of the aircraft.json file to feed the logic to create the simple text.

I'll play around with the idea a bit more and see where it goes.

feel free to close this issue if you want to keep the issues list clean (as this is not an issue, just an enhancement)

mutability, 2017-06-20:
https://github.com/mutability/dump1090-tools/blob/master/collectd/dump1090.py
may be a useful starting point for you

dernasherbrezon, 2017-09-22:
> Stats like this are already available as json, can you use that?

Saving stats to file every X seconds is not efficient. User might want to see them only once a year. Or less. This is especially critical for [raspberrypi deployments](https://github.com/dernasherbrezon/r2cloud), where disk is an SD card (slow and degrading over time). 

Can we add ```--stats-port``` configuration option which opens plain TCP port? Client could query this port as frequently as he likes to get json/CSV?

mutability, 2017-09-22:
You don't put the json dir on a sdcard, you put it on tmpfs.

The overhead of writing out stats is pretty trivial, AFAIK - can you show me some benchmarks showing a measurable slowdown due to it?

dernasherbrezon, 2017-09-22:
tmpfs will:
- complicate setup
- won't reduce unnecessary CPU usage

dump1090 already expose several ports. I thought it makes sense to export stats using the same approach.

Will try to come up with some metrics.

mutability, 2017-09-22:
The tmpfs approach is *exactly how dump1090 is already packaged*, it doesn't complicate things at all (most recent systems will have /run on tmpfs already)

I suggest you take a look at the network layer in dump1090 before going further down the custom TCP socket route - it is fairly horrible and there are strong reasons not to add anything more there. Also, you'd need to invent a protocol for that socket, and that protocol is going to look very like HTTP by the time you're done..

closed 2019-09-14 by mutability