mutability/dump1090 #203: Config for Apache 2.4

elmicha, 2017-06-23:
If you have an existing Apache 2.4 install and want to see dump1090 at http://localhost/dump1090, save the following lines to /etc/apache2/sites-available/dump1090.conf and run `sudo a2ensite dump1090` afterwards:

```
Alias "/dump1090/data" "/run/dump1090-mutability"
<Directory "/run/dump1090-mutability">
    Require all granted
</Directory>

Alias "/dump1090" "/usr/share/dump1090-mutability/html"
<Directory "/usr/share/dump1090-mutability/html">
    Options Indexes FollowSymLinks MultiViews
    AllowOverride None
    Require all granted
    DirectoryIndex gmap.html
</Directory>
```

E3V3A, 2017-06-30:
This didn't work for me. What exact steps did you take before?

elmicha, 2017-07-02:
After `dpkg-buildpackage -b` you need to install the package: 

    cd ..
    sudo dpkg -i dump1090-mutability_1.15~dev_amd64.deb

closed 2019-09-14 by mutability